mupq/pqm4 #11: Some implementations cannot be compiled using c89 standard

technotive, 2018-06-05:
I made a (now closed) issue(#6 ) on this earlier but the newly added qTesla128 also has some issues. Now the following two parts have issues:
- qTesla128
- Frodo640

Most errors amount to for-loops with internal variable declarations, but one issue in qTesla128 shows the following:
```C
fixed_llrint.c: In function ‘llrint_fixed’:
fixed_llrint.c:77:24: error: ‘__int32_t’ undeclared (first use in this function)
           if ((i0 & ~((__int32_t)1 << 31)) == 0)

```
Which is an unfamiliar issue to me. I replaced it with `int32_t` and that seems to hold up. I verified both the adjusted qTesla128 and Frodo640 code using test.py.
I have, however, no permission to push my changes to the branch I created locally. What would be the best way to provide my code? (My aim was to push it to a branch and create a pull request)

Proof of successful tests:
[image]

mkannwischer, 2018-06-05:
Thanks for your work. 
Please fork pqm4, push your changes there, and then create a pull request back into pqm4.

Ko-, 2018-06-05:
Most of the `ref` implementations are based on other people's code. Moreover, we never really claimed C89 compatibility. In particular, `stdint.h` is used in several places and that was only standardized with C99 anyway. The `__int32_t` is based on the newlib implementation of `llrint`. But yes, feel free to send a PR. :)

technotive, 2018-10-10:
I have a fix ready in my fork, I opted to add the `-std=c99` option. I will do a pull request once my fork is stable.

closed 2018-10-10 by technotive