mupq/pqm4 #17: Something wrong with the makefile

zdxxwqx, 2018-09-19:
When I type the "make" command, some errors are shown as follows:
Makefile:13: recipe for target 'frodo640.o' failed
make[1]: *** [frodo640.o] Error 1
make[1]: Leaving directory '/home/abc/Documents/pqm4/crypto_kem/frodo640-cshake/opt'
Makefile:84: recipe for target 'crypto_kem/frodo640-cshake/opt' failed
make: *** [crypto_kem/frodo640-cshake/opt] Error 2
What should I do to solve them?

zdxxwqx, 2018-09-19:
Well，I solved the problem, but when I run the "python3 hostside/host_unidirectional.py" command to receive the output, the terminal output the errors as follows:
Traceback (most recent call last):
  File "host_unidirectional.py", line 5, in <module>
    dev = serial.Serial("/dev/ttyUSB0", 115200)
  File "/usr/local/lib/python3.5/dist-packages/serial/serialutil.py", line 240, in __init__
    self.open()
  File "/usr/local/lib/python3.5/dist-packages/serial/serialposix.py", line 286, in open
    self._update_dtr_state()
  File "/usr/local/lib/python3.5/dist-packages/serial/serialposix.py", line 634, in _update_dtr_state
    fcntl.ioctl(self.fd, TIOCMBIS, TIOCM_DTR_str)
OSError: [Errno 5] Input/output error
How to solve it?

Ko-, 2018-09-19:
So `/dev/ttyUSB0` exists, but when setting the DTR (data terminal ready) bit, an OSError is thrown. I haven't seen this before, but it looks similar to [this](https://github.com/pyserial/pyserial/issues/67). Which version of pyserial are you using? I'm on 3.4. Could you try changing it to `serial.Serial("dev/ttyUSB0", 115200, rtscts=True, dsrdtr=True)`?

zdxxwqx, 2018-09-20:
OK，by changing the parameter of the Serial() the errors are no longer displayed. However, there aren't any returned data. Before run the "python3 host_unidirectional.py" I had write the crypto_kem_newhope1024cca_ref_speed.bin file into the board. The UART-USB connector that I use is the ftdi cable, and the outputs of the lsusb are as follows:
Bus 002 Device 005: ID 067b:2303 Prolific Technology, Inc. PL2303 Serial Port
Bus 002 Device 004: ID 0483:3748 STMicroelectronics ST-LINK/V2
So, what is the reason of this?

Ko-, 2018-09-20:
That's hard to see from my side. Which wire is connect to which pin? Did you try pressing the hardware reset button on the board while the Python script is running?

zdxxwqx, 2018-09-20:
I connencted TXD→RX and RXD→TX（from cable to board）I had ran a basic program which is about the serial port printing tests and I read the output in Windows10. I also pressed the RST button but it still without any returned data.

zdxxwqx, 2018-09-20:
Just now I run that basic program again and I am able to read the output with the host_unidirectional.py. But when I run the newhope there isn't any output.

zdxxwqx, 2018-09-20:
I want to execute newhope benchmark test and I run the command "python3 test.py newhope1024cca sphincs-shake256-128s". But the outputs are sa follows:
Flashing bin/crypto_sign_sphincs-shake256-128s_ref_test.bin..
Flashed, now running tests...
timed out while waiting for the markers
What is the reason that the program timed out? Is this related to the problem of no output? And what is the meaning of the parameter "sphincs-shake256-128s"

zdxxwqx, 2018-09-29:
Thank you for your inquiry. After seeing your board, I found that there is a huge difference between our devices. I also tried to connect the PA2 and PA3 pins of my board, but I still can't receive the output. This result is within my expectation. Since I am a beginner in stm32, there is still a lot to learn. Next I plan to modify your program based on my device. By the way, what is the model number of your board? And where can I buy it? I still want to run this program. Thank you again for your concern.

Ko-, 2018-09-29:
It's an [STM32F407 discovery board](https://www.st.com/en/evaluation-tools/stm32f4discovery.html). What device do you have? Maybe there's not that much that needs to change.

jowlo, 2018-09-29:
I do not want to hijack this issue, but here's a side note:

While accidentally all these 2-wire wirings might work because both USB devices (UART-Adapter and Board) use the USB Controller's ground when connected to the same computer, you should definitely connect the two grounds on the board and the adapter.

The serial connection essentially is a slimmed down (aka **3**-wire) version of [RS-232](https://en.wikipedia.org/wiki/RS-232). Having no (or flaky) common ground potential can lead to corrupted or no data transmitted at all.

zdxxwqx, 2018-10-02:
My device is STM32F429, and since I had only connect the PA2 and PA3 pins last time, my chip ID can't be recognized. Is this related to the problem that Jowlo said?

Ko-, 2018-10-02:
What do you mean by "chip ID can't be recognized"?

I don't think it's related to improper grounding, although it can't hurt to try. My guess is that it's either related to the linker script that we use or to the specific clock frequencies that we set. But I'm afraid we can't really test anything as none of us has an STM32F429.

zdxxwqx, 2018-10-03:
The "chip ID can't be recognized" means I can't connect the board with st-link and I can't download the bin file to the board. Since the last time I connected the PA2 and PA3 pains and call the test.py. I couldn't find the reason, maybe the flash of my board is too small causing the downloaded program to cover the area related to chip identification. So, I plan to buy an STM32F407 discovery board to repeat this.

Ko-, 2018-10-10:
The STM32F429 should have the same or more flash memory, depending on which model exactly, so I don't think that's it. Anyway, I'm closing this issue for now.

closed 2018-10-10 by Ko-