mupq/pqm4 #21: .m4ignore does not work as indicated in readme

technotive, 2018-10-10:
> If the implementation you're adding is such a host-side-only reference implementation, place a file called `.m4ignore` in the subdirectory containing the implementation. In that case the Makefile is not required to contain the `libpqm4` target.

I am currently adjusting a project using __uint128_t in the reference version, and the functionality mentioned would be beneficial. The compiler gets angry when compiling the ref for the m4. However, the make script seems to still want the `libpqm4.a` target.

Ko-, 2018-10-10:
You're right, that description is not entirely accurate. The build system is currently agnostic of the `.m4ignore` files and if it's a `ref` folder, it expects a `libpqm4.a` target in the Makefile in that directory. Only the Python scripts correctly ignore these binaries. A quick workaround is to keep your `libpqm4.a` target and have it create some `libpqm4.a` file.

@mkannwischer @joostrijneveld @cryptojedi Do we want to update the Makefile for this?

mkannwischer, 2018-10-10:
We need to clean up some stuff in the Makefile anyway, so I'm in favor of also fixing this at the same time.

closed 2018-10-11 by mkannwischer

joostrijneveld, 2018-10-11:
I agree that the Makefile needs a more serious cleaning, but this seemed like a small enough and sufficiently isolated issue to fix separately.

PR #22 (2018-10-11): Check for .m4ignore in Makefile
In particular, this prevents breaking with compiler complaints when trying to build all schemes using a simple `make`. There is still quite some duplication in the Makefile, but we can resolve that in a different changeset.

Closes #21