mupq/pqm4 #29: Measure time spent in SHA2, SHA3, and randombytes

mkannwischer, 2018-10-22:
As in our recent [paper](https://kannwischer.eu/papers/2018_polymul-z2mx-m4_preprint20181019.pdf), it would be nice to have some profiling information on how much time is spent in hashing and `randombytes`. 
This should also give an intuition how the schemes sample randomness. 

We implemented this by adding/subtracting the current cycle count to a global variable when entering/leaving the respective function. This should be done only for profiling, but not in the benchmarks since it can have a huge performance impact. It would be nice to port that to PQM4.

In case we want to use `systick` for this, then we will run into the problem that if the overflow occurs within one of the profiled functions the cycle counts are useless. In our paper experiments, we just detected this case and ran it several times until we had no overflow there - this won't work for SPHINCS.

Ko-, 2018-10-22:
I don't really see why we couldn't account for that. We're already doing that for counting cycles in `speed.c` with `systick`, right? Every overflow triggers an interrupt and those are counted. Of course this logic has some overhead, but if it only happens every 2.400.000 cycles then that shouldn't be very significant.

There's also [this approach](https://mcuoneclipse.com/2015/08/23/tutorial-using-gnu-profiling-gprof-with-arm-cortex-m/) to profiling applications with `gprof`. I looked into it at some point, but it's slightly more cumbersome. Definitely a downside is that it requires extra hardware, although I think we have that in the office. It's probably technically possible to do without, but that would require implementing some low-level I/O functions ourselves to make sure that `gprof.out` gets written to USART on exit.

mkannwischer, 2018-10-23:
On Mon, Oct 22, 2018 at 03:10:41PM -0700, Ko- wrote:
> I don't really see why we couldn't account for that. We're already doing that for counting cycles in `speed.c` with `systick`, right? Every overflow triggers an interrupt and those are counted. Of [...]

You are absolutely right - we just need to check if there was an overflow at the end of each call.

commit d067bca8 (2019-03-21): adds hashing benchmarks. Closes #29
commit 8627f530 (2019-05-01): pqm4 V2 (#78)

* Add mupq submodule

* Word-align the address of polymul_asm (#37)

* added hal.h

* hal.h

* Rewrite utils to use interface.py

Fix dev references

* Actually register submodule

* add stm32f4 hal.c

* Refactor Makefile; build single binaries

* kick-out schemes that did not make it to round2. Resolves #40

* Shrink RAM from 128 KiB to 112 KiB (#73)

By not using SRAM2 we get more consistent benchmarks since SRAM2 seems to slow
down memory accesses. This lead to schemes that are wasteful with stack
(i.e., overflowing the 16 KiB SRAM2 into SRAM1) were faster.

* Move common sou [... cut]
closed 2019-05-01 by mkannwischer