mupq/pqm4 #55: Integrate BIKE [Round2]

mkannwischer, 2019-03-13:
if possible - otherwise document why not.

mkannwischer, 2019-06-12:
We are working on integration into PQClean (without libraries). Let's wait for that.

Ko-, 2019-08-24:
BIKE has updated [reference implementations](https://bikesuite.org/files/round2/ref-impl/Reference_Implementation-2019.06.30.1.zip), although they still use the NTL and GMP libraries. On my laptop, `bike-demo-test` for `BIKE-1-CCA` at level 1 has a peak memory consumption of 246.3 KiB, which is not going to fit on our platform. I have not yet checked the optimized implementations.

mjosaarinen, 2020-02-19:
BIKE has just updated their specs to the third round and to use the standard SHA-384 and I've written an implementation that doesn't require NTL or GMP and actually supports all variants. So should be able to contribute an implementation if I can just find little time to adjust it for PQM4.
I'll have to double-check the memory requirement but it's a lot less than what is used to be.

Ko-, 2020-05-02:
The spec was updated again to v4.0. Just one variant with two security levels now. The reference implementation still uses OpenSSL and NTL. @mjosaarinen any thoughts on whether your adaptions to not depend on external libraries directly carry over to v4.0?

mkannwischer, 2021-02-18:
Fiiiiiinaaallly resolved by #175. 
Thank you @devillegna!!

closed 2021-02-18 by mkannwischer