mupq/pqm4 #57: Integrate HQC [Round2]

mkannwischer, 2019-03-13:
if possible - otherwise document why not.

Ko-, 2019-05-14:
This scheme might fit. Unfortunately the implementations depend on the `openssl`, `gmp`, `ntl`, and `gf2x` libraries [[spec](http://pqc-hqc.org/doc/hqc-specification_2019-04-10.pdf)].

mkannwischer, 2019-06-12:
We are working on integration into PQClean (without libraries). Let's wait for that.

mkannwischer, 2019-06-25:
https://github.com/PQClean/PQClean/pull/202

Ko-, 2019-08-24:
HQC has updated [reference implementations](https://pqc-hqc.org/doc/hqc-reference-implementation_2019-08-24.zip), although they still depend on the same libraries. `hqc-128-1` has a peak memory consumption of 198.4 KiB on my laptop, which is not going to fit on our platform. I have not yet checked the optimized implementations.

Ko-, 2019-08-24:
Apparently, [optimized](https://pqc-hqc.org/doc/hqc-optimized-implementation_2019-08-24.zip) implies AVX2 here, so that's not very useful to `pqm4`.

commit 35991c6e (2020-12-01): add HQC benchmarks. Closes #57
commit 157e2717 (2020-12-09): Update PQClean (#172)

* Updated sampling of uniform matrix of Kyber to round-3 tweaked approach

* Integrated changes to noise sampling in Kyber512

* Updated links in kyber512-90s/m4 to use round-3 noise generation of kyber512/m4

* source ntruprime from pqclean

* https://github.com/PQClean/PQClean/pull/324

* add HQC benchmarks. Closes #57

* Port https://github.com/PQClean/PQClean/pull/337

This ports a fix from upstream. This does not change performance by more than a few cycles.
For details see
https://github.com/jschanck/ntru/commit/e0ab9525f1797dcff875c67f08f56db03f3c7deb

* Port http [... cut]
closed 2020-12-09 by Ko-