mupq/pqm4 #61: Integrate ROLLO [Round2]

mkannwischer, 2019-03-13:
if possible - otherwise document why not.

Ko-, 2019-05-07:
This scheme might fit. Unfortunately the implementations depend on the `openssl`, `gmp`, and `ntl` libraries.

mkannwischer, 2019-06-12:
We are working on integration into PQClean (without libraries). Let's wait for that.

Ko-, 2019-08-24:
ROLLO has updated [reference](https://pqc-rollo.org/doc/rollo-reference-implementation_2019-08-24.zip) and [optimized](https://pqc-rollo.org/doc/rollo-optimized-implementation_2019-08-24.zip) implementations, where dependencies on external libraries appear to be removed. I briefly tried the reference implementation of `ROLLO-I-128` and its peak memory consumption is 17.38 KiB. For the largest parameter set, `ROLLO-III-256`, it is only 44.90 KiB, so it looks like we have another scheme to add!

PR #120 (2019-10-08): Add ROLLO-{I,II,III}-{128,192,256} ref implementations
This adds the [updated reference implementations](https://pqc-rollo.org/doc/rollo-reference-implementation_2019-08-24.zip) for all parameter sets of ROLLO. Also fixes #61.
closed 2019-10-08 by mkannwischer