mupq/pqm4 #63: Integrate RQC [Round2]

mkannwischer, 2019-03-13:
if possible - otherwise document why not.

Ko-, 2019-05-07:
This scheme might fit. Unfortunately the implementations depend on the `openssl`, `gmp`, `ntl`, and `gf2x` libraries [[spec](http://pqc-rqc.org/doc/rqc-specification_2019-04-10.pdf)].

mkannwischer, 2019-06-12:
We are working on integration into PQClean (without libraries). Let's wait for that.

Ko-, 2019-08-24:
RQC has updated [reference](https://pqc-rqc.org/doc/rqc-reference-implementation_2019-08-24.zip) and [optimized](https://pqc-rqc.org/doc/rqc-optimized-implementation_2019-08-24.zip) implementations. They removed all dependencies on external libraries. I briefly tested the reference implementation of `rqc128` on my laptop, and it only has a peak memory consumption of 30.56 KiB, so it looks like we have a new scheme to add!

PR #121 (2019-10-08): Add RQC{128,192,256} ref implementations
This adds the [updated reference implementations](http://pqc-rqc.org/doc/rqc-reference-implementation_2019-08-24.zip) of all parameter sets of RQC. Some changes were made to get rid of x64 assembly code, to address some warnings, and to get rid of a memory leak in decapsulation.

Fixes #63.
closed 2019-10-08 by mkannwischer