mupq/pqm4 #66: Integrate GeMSS [Round2]

mkannwischer, 2019-03-13:
if possible - otherwise document why not.

Ko-, 2019-05-07:
Why not: these public key sizes will never fit. [[NIST submission](https://csrc.nist.gov/CSRC/media/Projects/Post-Quantum-Cryptography/documents/round-2/submissions/GeMSS-Round2.zip)]

* GeMSS128: 352.19 kB
* BlueGeMSS128: 363.61 kB
* RedGeMSS128: 375.21 kB
* GeMSS192: 1237.96 kB
* BlueGeMSS192: 1264.12 kB
* RedGeMSS192: 1290.54 kB
* GeMSS256: 3040.70 kB
* BlueGeMSS256: 3087.96 kB
* RedGeMSS256: 3135.59 kB

closed 2019-05-09 by mkannwischer

Ko-, 2020-04-15:
Today GeMSS updated their spec and implementation. Table 1 of the [changes document](https://www-polsys.lip6.fr/Links/NIST/changes_round2_V2.pdf) still lists the same public key sizes. If I understand it correctly, that is without bit packing, which is also proposed in Section 2.3 as a technique to compress the public key. There it says "we save up to 18% of the public-key size", so I believe that GeMSS will still not fit on the stm32f412, even with bit packing.