mupq/pqm4 #68: Integrate MQDSS [Round2]

mkannwischer, 2019-03-13:
if possible - otherwise document why not.

mkannwischer, 2019-05-16:
I've tested the stack usage on the host (including keys, signature, and message)
- `mqdss-48`: 201,224 bytes
- `mqdss-64`:  423,512 bytes 

This will never fit entirely. `mqdss-48` might fit if you don't do key generation on the device itself.

closed 2019-05-16 by mkannwischer

joostrijneveld, 2019-05-17:
Initially I thought F was the bottleneck, but also the allocations of r0, r1, t0, t1, e0, e1 and gx add up quite a bit. If someone really wants this, it's possible to write this in a way that these values are computed more incrementally, but that's some work. It's important to consider that r0, t0 and e0 are required after computing the challenge, and would need to be recomputed (i.e. expanded from the seed).

Similarly, the commitments are currently hashed all at once; that could be incremental as well.