mupq/pqm4 #69: Integrate Picnic [Round2]

mkannwischer, 2019-03-13:
if possible - otherwise document why not.

Ko-, 2019-05-15:
In Table 16 and Table 17 of the [The Picnic Signature Algorithm Design Document, version 2.0](https://github.com/microsoft/Picnic/raw/master/spec/design-v2.0.pdf), valgrind is used to measure peak memory usage (stack + heap) of the reference and optimized implementations, respectively, for all 9 parameter sets of Picnic.

> When creating a signature, peak memory usage ranged from about 190K to 33M bytes, as shown in Table 16, while verification ranged from about 132K to 32M bytes.

This will never fit.

closed 2019-05-16 by mkannwischer

Ko-, 2020-04-15:
Today the Picnic team announced v3.0 of [their spec](https://github.com/microsoft/Picnic/blob/master/spec/spec-v3.0.pdf), which contains new parameter sets. Unfortunately, the peak memory consumption measurements in the design document have not been extended to these new parameter sets. (Can't find it in [design v2.2](https://github.com/microsoft/Picnic/raw/master/spec/design-v2.2.pdf) or the [paper](https://github.com/microsoft/Picnic/raw/master/spec/picnic3-eprint.pdf).) So, while I suspect the memory requirements to lie around the same ballpark, this remains to be verified.