mupq/pqm4 #71: Namspacing

mkannwischer, 2019-03-19:
Some people showed interest in bundling more than one scheme from pqm4 together which means that we would need to namespace everything. 

Since we have this in https://github.com/PQClean/PQClean/ anyway it should be straightforward to just keep it. 

Any strong opinions?

Ko-, 2019-03-19:
I don't immediately see the use case for bundling multiple schemes that are not the combination of a KEM and a signature, but sure, if there's a demand for this, let's do it. [This](https://stackoverflow.com/questions/389827/namespaces-in-c) lists some alternative namespacing implementations. I think I find the `SHORT_NAMES` solution more elegant than the `##` stuff in PQClean, but I don't have a strong preference. I guess it has the downside of having to do some more changes to all the schemes, instead of being able to keep the ugly stuff in the central files (`speed.c`, etc.). For that reason I'm not a big fan of the global name struct solution.

cryptojedi, 2019-03-21:
Ko- <notifications@github.com> wrote:
> I don't immediately see the use case for bundling multiple schemes
> that are not the combination of a KEM and a signature, but sure, if
> there's a demand for this, let's do it. 

One reason would be to sign documents with multiple signatures to
protect against one of the schemes being broken.

mkannwischer, 2019-11-15:
I'm closing this for now. 
If implementations need to be namespaced, we cannot symlink across parameter sets. I think this outweighs the benefit of being able to link multiple schemes together for pqm4.

closed 2019-11-15 by mkannwischer