mupq/pqm4 #104: _mm_aesenclast_si128 implementation

liorko87, 2019-08-04:
Hello,

Is there an implementation for `_mm_aesenclast_si128` instruction? The instruction is part of AES(Advanced Encryption Standard) [instruction set](https://software.intel.com/sites/landingpage/IntrinsicsGuide/#text=_mm_aesenclast_si128&expand=236).

Lior

mkannwischer, 2019-08-04:
I think I don't quite get your question. 
Yes, we do have an ARMv7-M assembly implementation of AES:
https://github.com/mupq/pqm4/blob/master/common/aes.S

It is the one by Schwabe and Stoffelen described in https://eprint.iacr.org/2016/714.pdf.

Obviously, this implementation also implements the last round: 
https://github.com/mupq/pqm4/blob/93d444af0f5931025b7b1b15e66f08f1d6300ded/common/aes.S#L481

If you only want the implementation of the last round, you could rip it out and expose it as a separate function, although it is unclear to me why one would want to do that.

closed 2019-08-04 by mkannwischer

Ko-, 2019-08-04:
`_mm_aesenclast_si128` is an intrinsic that most compilers implement when they are targeting **Intel** processors with the **AES-NI** instruction set. This targets **ARM** processors, so that works differently. The STM32F407 chip does not have a cryptographic coprocessor, so we use a software implementation of AES. For other ST chips that do have a cryptographic coprocessor or even for ARMv8-A processors, you would still not use the same intrinsics as you would on Intel.