mupq/pqm4 #134: new version of round5 -- cca kems official etc

mjosaarinen, 2020-02-19:
Hello PQM4!

The Round5 team is announcing "third NIST round roadmap" this week, and that includes some changes such as parameter naming changes. The CCA KEMs are finally official external parts of the system -- the PQM4 implementation diverged a bit from the specification in this aspect.

I've updated the embedded implementation at https://github.com/r5embed/r5embed to reflect changes. The PKE (AES-GCM) variants are gone from that implementation, as are the SNEIK variants. The KEMs are in sync with the new reference code at https://github.com/round5/code

It would be great if these changes can be adopted to PQM4 as well. The code should be actually easier to adopt than the previous one due to the abovementioned issues w.r.t CCA KEMs. The big pull was https://github.com/r5embed/r5embed/commit/45ddd289b34942701bf4d222346e43f19a75a714 but it may be easier to just copy the files.

Cheers,
- markku

Ko-, 2020-03-01:
Hi Markku,

Thanks for the heads-up! I'll integrate the updated implementation.

Cheers,
Ko

dsprenkels, 2020-03-05:
@Ko-, I ran the benchmarks (https://github.com/mupq/pqm4/commit/c5b4792d27245b4f6fade7227593e5e4038820d5).

Ko-, 2020-03-08:
Thanks a lot @dsprenkels! But it looks like the `opt` benchmarks are disappearing with that commit...

dsprenkels, 2020-03-11:
@Ko-, I will take a look at it.

Ko-, 2020-03-11:
Only if it's not too much trouble. :) I ordered a board for myself last Monday, so soonish I should be able to run benchmarks again.

Ko-, 2020-03-25:
Sorry that it took so long, but I finally integrated the new Round5 implementations.

PR #135 (2020-03-25): Update Round5 to third round
Fixes #134.
closed 2020-03-25 by Ko-