mupq/pqm4 #145: GCC 9.3.0

mkannwischer, 2020-04-16:
Recently, gcc 9.3.0 was released and landed on the  [Arch arm-none-eabi-gcc/](https://www.archlinux.org/packages/community/x86_64/arm-none-eabi-gcc/).

We were wondering if this impacts performance at all, so @dsprenkels ran some benchmarks: 
[gcc9.3.0benchmarks.txt](https://github.com/mupq/pqm4/files/4484490/gcc9.3.0benchmarks.txt)

Note that we only did one execution so far, so the results for schemes where the run-time differs a lot per execution like Dilithium or Falcon should be ignored. 
The speed-ups are as follows: 

# KEMs 

| scheme | impl | speedup keygen | speedup encaps | speed-up decaps | 
| --- | --- | --- | --- | --- |
| babybear (1 executions) | clean | -0.00% | -0.00% | 0.00% |
| babybear (1 executions) | opt | -9.51% | -9.06% | -10.97% |
| babybear-ephem (1 executions) | clean | -0.00% | -0.00% | 0.00% |
| babybear-ephem (1 executions) | opt | -9.51% | -10.40% | -19.97% |
| firesaber (1 executions) | clean | -0.06% | -0.06% | -0.05% |
| firesaber (1 executions) | m4 | -0.04% | 0.20% | 0.17% |
| frodokem640aes (1 executions) | m4 | -2.82% | -2.82% | -2.88% |
| frodokem640shake (1 executions) | m4 | -0.00% | -0.00% | 0.01% |
| frodokem640shake (1 executions) | opt | -0.00% | -0.00% | 0.00% |
| kyber1024 (1 executions) | clean | -0.52% | -0.25% | -0.59% |
| kyber1024 (1 executions) | m4 | 0.01% | 0.01% | 0.02% |
| kyber1024-90s (1 executions) | clean | -0.01% | -0.01% | 0.00% |
| kyber1024-90s (1 executions) | m4 | 0.00% | 0.01% | 0.02% |
| kyber512 (1 executions) | clean | -0.97% | -0.50% | -0.94% |
| kyber512 (1 executions) | m4 | -0.05% | -0.02% | -0.01% |
| kyber512-90s (1 executions) | clean | -0.06% | -0.03% | -0.01% |
| kyber512-90s (1 executions) | m4 | -0.05% | -0.02% | 0.01% |
| kyber768 (1 executions) | clean | -0.69% | -0.35% | -0.75% |
| kyber768 (1 executions) | m4 | -0.03% | -0.02% | -0.01% |
| kyber768-90s (1 executions) | clean | -0.02% | -0.00% | 0.00% |
| kyber768-90s (1 executions) | m4 | -0.07% | -0.05% | -0.03% |
| lac128 (1 executions) | ref | -0.15% | -0.17% | -0.11% |
| lac192 (1 executions) | ref | -0.12% | -0.06% | -0.03% |
| lac256 (1 executions) | ref | -0.06% | -0.07% | -0.04% |
| lightsaber (1 executions) | clean | -0.08% | -0.06% | -0.05% |
| lightsaber (1 executions) | m4 | 0.08% | 0.06% | 0.10% |
| mamabear (1 executions) | clean | -0.00% | -0.00% | 0.00% |
| mamabear (1 executions) | opt | -6.59% | -6.42% | -8.70% |
| mamabear-ephem (1 executions) | clean | -0.00% | -0.00% | 0.00% |
| mamabear-ephem (1 executions) | opt | -6.52% | -7.11% | -18.01% |
| newhope1024cca (1 executions) | clean | 0.63% | 0.66% | 0.61% |
| newhope1024cca (1 executions) | m4 | -0.03% | -0.02% | -0.01% |
| newhope1024cpa (1 executions) | clean | 0.73% | 0.77% | 0.07% |
| newhope1024cpa (1 executions) | m4 | -0.01% | -0.01% | 0.00% |
| newhope512cca (1 executions) | clean | 0.69% | 0.73% | 0.66% |
| newhope512cca (1 executions) | m4 | -0.04% | -0.01% | 0.00% |
| newhope512cpa (1 executions) | clean | 0.72% | 0.77% | 0.13% |
| newhope512cpa (1 executions) | m4 | -0.03% | -0.02% | 0.00% |
| ntruhps2048509 (1 executions) | clean | -0.48% | -0.26% | -0.04% |
| ntruhps2048509 (1 executions) | m4 | 0.00% | -1.18% | -0.61% |
| ntruhps2048677 (1 executions) | clean | -0.93% | -0.21% | -0.03% |
| ntruhps2048677 (1 executions) | m4 | 0.32% | -1.06% | -0.52% |
| ntruhps4096821 (1 executions) | clean | -0.24% | -0.17% | -0.03% |
| ntruhps4096821 (1 executions) | m4 | -0.00% | -1.02% | -0.50% |
| ntruhrss701 (1 executions) | clean | -0.23% | -0.10% | -0.03% |
| ntruhrss701 (1 executions) | m4 | -0.00% | -1.13% | -0.55% |
| ntrulpr653 (1 executions) | ref | -0.03% | -0.01% | -0.01% |
| ntrulpr761 (1 executions) | ref | -0.02% | -0.01% | -0.01% |
| ntrulpr857 (1 executions) | ref | -0.02% | -0.01% | -0.00% |
| papabear (1 executions) | clean | -0.00% | -0.00% | 0.00% |
| papabear (1 executions) | opt | -4.98% | -4.90% | -7.15% |
| papabear-ephem (1 executions) | clean | -0.00% | -0.00% | 0.00% |
| papabear-ephem (1 executions) | opt | -4.94% | -5.34% | -16.77% |
| r5n1-1cca-0d (1 executions) | m4 | 0.29% | 0.36% | 0.33% |
| r5n1-1cca-0d (1 executions) | opt | 0.22% | 0.29% | 0.23% |
| r5n1-1cpa-0d (1 executions) | m4 | -0.03% | -0.55% | 0.27% |
| r5n1-1cpa-0d (1 executions) | opt | 0.08% | -0.07% | 0.42% |
| r5n1-3cca-0d (1 executions) | m4 | -0.12% | 0.02% | -0.14% |
[... 97 lines cut ...]
| sphincs-sha256-192f-simple (1 executions) | clean | -0.03% | 0.04% | -1.12% |
| sphincs-sha256-192s-robust (1 executions) | clean | -0.19% | -0.18% | -7.41% |
| sphincs-sha256-192s-simple (1 executions) | clean | -0.03% | -0.08% | 0.35% |
| sphincs-sha256-256f-robust (1 executions) | clean | -0.15% | -0.11% | -1.35% |
| sphincs-sha256-256f-simple (1 executions) | clean | -0.05% | -0.08% | 2.06% |
| sphincs-sha256-256s-robust (1 executions) | clean | -0.16% | -0.14% | 0.13% |
| sphincs-sha256-256s-simple (1 executions) | clean | 0.01% | -0.05% | 2.69% |
| sphincs-shake256-128f-robust (1 executions) | clean | 1.52% | 1.41% | 3.74% |
| sphincs-shake256-128f-simple (1 executions) | clean | 1.50% | 1.75% | 3.46% |
| sphincs-shake256-128s-robust (1 executions) | clean | 1.52% | 1.52% | 3.37% |
| sphincs-shake256-128s-simple (1 executions) | clean | 1.50% | 1.49% | 4.97% |
| sphincs-shake256-192f-robust (1 executions) | clean | 1.52% | 1.57% | -0.46% |
| sphincs-shake256-192f-simple (1 executions) | clean | 1.50% | 1.33% | 3.67% |
| sphincs-shake256-192s-robust (1 executions) | clean | 1.52% | 1.53% | 4.83% |
| sphincs-shake256-192s-simple (1 executions) | clean | 1.50% | 1.48% | 6.73% |
| sphincs-shake256-256f-robust (1 executions) | clean | 1.51% | 1.53% | 2.77% |
| sphincs-shake256-256f-simple (1 executions) | clean | 1.50% | 1.50% | 1.96% |
| sphincs-shake256-256s-robust (1 executions) | clean | 1.50% | 1.51% | -1.51% |
| sphincs-shake256-256s-simple (1 executions) | clean | 1.50% | 1.49% | -2.53% |

Somehow the optimized implementations of the ThreeBears seem to suffer a lot from this update, whereas the SHAKE parameter sets of SPHINCS+ get a small speed-up. 
Besides that, I think the changes are negligible. 
Any thoughts on what we should do to the benchmarks? How about updating that changed by more than 1%? 

Compilers are amazing.

dsprenkels, 2020-04-16:
Currently in the process of running this command:

```sh
for i in $(seq 99); do printf 'ITERATION: %d' $i; ./benchmarks.py --nosize --nostack babybear babybear-ephem firesaber frodokem640aes frodokem640shake kyber1024 kyber1024-90s kyber512 kyber512-90s kyber768 kyber768-90s lac128 lac192 lac256 lightsaber mamabear mamabear-ephem newhope1024cca newhope1024cpa newhope512cca newhope512cpa ntruhps2048509 ntruhps2048677 ntruhps4096821 ntruhrss701 ntrulpr653 ntrulpr761 ntrulpr857 papabear papabear-ephem r5n1-1cca-0d r5n1-1cpa-0d r5n1-3cca-0d r5n1-3cpa-0d r5n1-5cca-0d r5n1-5cpa-0d r5nd-1cca-0d r5nd-1cca-5d r5nd-1cpa-0d r5nd-1cpa-5d r5nd-3cca-0d r5nd-3cca-5d r5nd-3cpa-0d r5nd-3cpa-5d r5nd-5cca-0d r5nd-5cca-5d r5nd-5cpa-0d r5nd-5cpa-5d rollo-I-128 rollo-I-192 rollo-I-256 rollo-II-128 rollo-II-192 rollo-II-256 rollo-III-128 rollo-III-192 rollo-III-256 rqc128 rqc192 rqc256 saber sikep434 sikep503 sikep610 sikep751 sntrup653 sntrup761 sntrup857 dilithium2 dilithium3 dilithium4 falcon-1024 falcon-512 falcon-512-tree luov-47-42-182-chacha luov-47-42-182-keccak luov-61-60-261-chacha luov-61-60-261-keccak luov-7-57-197-chacha luov-7-57-197-keccak; done
```

mkannwischer, 2020-05-29:
We now have GCC10 benchmarks #156

closed 2020-05-29 by mkannwischer