mupq/pqm4 #152: Update HQC

Ko-, 2020-04-22:
The HQC team announced an [updated spec](https://pqc-hqc.org/doc/hqc-specification_2020-04-21.pdf) with smaller parameters, new additional parameter sets called HQC-RMRS and their implementations are now constant-time. Currently the HQC implementations are only in PQClean and all in the `skip_list`, so then nothing would change for pqm4, but with smaller parameters and HQC-RMRS appearing to be smaller than HQC, I don't know whether these will fit or not. And whether this should be updated in PQClean or in mupq. Their new reference code lives [here](https://pqc-hqc.org/doc/hqc-reference-implementation_2020-04-21.zip).

thomwiggers, 2020-04-22:
See also https://github.com/PQClean/PQClean/issues/291

Ko-, 2020-05-30:
The implementation was updated again. This time supposedly really constant-time. [Specification](http://pqc-hqc.org/doc/hqc-specification_2020-05-29.pdf) and [reference code](https://pqc-hqc.org/doc/hqc-reference-implementation_2020-05-29.zip).

mkannwischer, 2020-12-09:
Solved by #172

closed 2020-12-09 by Ko-