mupq/pqm4 #154: Error while Building Binaries

adithyabhatkajake, 2020-04-27:
Hi, when running `python build_everything.py` while following the build instructions provided in `Readme.md`, I get the following error:

```
/usr/lib/gcc/arm-none-eabi/9.3.0/../../../../arm-none-eabi/bin/ld: /tmp/cctQz4u1.o: in function `ZKeyGen':
kem.c:(.text+0x20): undefined reference to `aes256_keyexp'
/usr/lib/gcc/arm-none-eabi/9.3.0/../../../../arm-none-eabi/bin/ld: /tmp/cctQz4u1.o: in function `ZEncrypt':
kem.c:(.text+0x1fa): undefined reference to `aes256_keyexp'
/usr/lib/gcc/arm-none-eabi/9.3.0/../../../../arm-none-eabi/bin/ld: kem.c:(.text+0x2f8): undefined reference to `aes256_keyexp'
collect2: error: ld returned 1 exit status
make: *** [Makefile:86: elf/crypto_kem_ntrulpr761_m4f_test.elf] Error 1
```

I use Arch Linux and have installed all the dependencies. This can be reproduced by running
` make IMPLEMENTATION_PATH=crypto_kem/ntrulpr761/m4f bin/crypto_kem_ntrulpr761_m4f_test.bin`
I have attached a detailed log of the errors here [errors.log](https://github.com/mupq/pqm4/files/4541681/errors.log)

Is there any way I can get this working? Am I missing something?

Ko-, 2020-04-27:
Hi, thanks for reporting this! You're not missing anything, we missed something. That code was made to work before pull request #148 that split `aes256_keyexp` into `aes256_ctr_keyexp` and `aes256_ecb_keyexp`. I just pushed a fix (479f4a5). Can you pull those changes and try again? I can compile them now.

adithyabhatkajake, 2020-04-27:
This time there's an error while building `lac-128-v3a` when linking `pseudo_random_bytes`.

Errors: [errors.log](https://github.com/mupq/pqm4/files/4541981/errors.log)

mkannwischer, 2020-04-27:
Same problem, #147 didn't include the changes from #148. 
I must have been asleep when merging these changes. Sorry for the inconvience. 
I fixed it in https://github.com/mupq/mupq/commit/8fd19417bf12ae5cdfa6f7d2649bf770a2b466dc /  https://github.com/mupq/pqm4/commit/a1bbbd9c541b972740ee1a1fba6d80958a916927

Can you `git pull --recurse-submodules` please and try again? I verified that `./build_everything.py` does not throw any errors anymore, so I hope you won't experience any further  issues with that.

adithyabhatkajake, 2020-04-28:
Thanks for the quick response everyone. 
I managed to build everything with this version and everything works.

closed 2020-04-28 by adithyabhatkajake