mupq/pqm4 #160: build error

marco-palumbi, 2020-06-15:
After last commit, running 

`python3 benchmarks.py kyber768`

I get this error:

```
INFO:Implementation:Building kyber768 - m4 - size
make: *** No rule to make target 'obj/crypto_kem_kyber768_m4_fastbasemul.o'.  Stop.
Traceback (most recent call last):
  File "benchmarks.py", line 23, in <module>
    test.test_all(sys.argv[1:])
  File "/home/marco/Shared/pqm4/mupq/mupq.py", line 248, in test_all
    if self.run_test(implementation) == -1:
  File "/home/marco/Shared/pqm4/mupq/mupq.py", line 299, in run_test
    implementation.build_objects(self.test_type)
  File "/home/marco/Shared/pqm4/mupq/mupq.py", line 85, in build_objects
    self.run_make(object_file)
  File "/home/marco/Shared/pqm4/mupq/mupq.py", line 67, in run_make
    return subprocess.check_call(makeflags)
  File "/usr/lib/python3.8/subprocess.py", line 364, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['make', 'IMPLEMENTATION_PATH=crypto_kem/kyber768/m4', 'obj/crypto_kem_kyber768_m4_fastbasemul.o']' returned non-zero exit status 2.
```

probably the last changes to Makefile are the culprits

Ko-, 2020-06-15:
Hi, thanks for pointing that out! You're completely right. Perhaps @mkannwischer hadn't had his coffee yet or something. ;) I reverted that in c32bcd0, so if you pull it should work again.

closed 2020-06-15 by Ko-

marco-palumbi, 2020-06-15:
thank you!
now it work again

mkannwischer, 2020-06-16:
Thanks, @marco-palumbi for reporting; and thanks @Ko- for fixing.
Indeed, an insufficient level of caffeine in my bloodstream is the only explanation I have for this.
My apologies.