mupq/pqm4 #164: building error

yusimple, 2020-10-27:
Makefile:85: recipe for target 'elf/crypto_kem_sikep503_m4_test.elf' failed
make: *** [elf/crypto_kem_sikep503_m4_test.elf] Error 1
Traceback (most recent call last):
  File "build_everything.py", line 12, in <module>
    mupq.BuildAll(M4Settings()).test_all(sys.argv[1:])
  File "/home/yuleitao/Documents/pqm4/mupq/mupq.py", line 248, in test_all
    if self.run_test(implementation) == -1:
  File "/home/yuleitao/Documents/pqm4/mupq/mupq.py", line 387, in run_test
    implementation.build_binary(test_type)
  File "/home/yuleitao/Documents/pqm4/mupq/mupq.py", line 74, in build_binary
    self.run_make(self.get_binary_path(type_))
  File "/home/yuleitao/Documents/pqm4/mupq/mupq.py", line 67, in run_make
    return subprocess.check_call(makeflags)
  File "/usr/local/lib/python3.6/subprocess.py", line 311, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['make', 'IMPLEMENTATION_PATH=crypto_kem/sikep503/m4', 'bin/crypto_kem_sikep503_m4_test.bin']' returned non-zero exit status 2.

yusimple, 2020-10-27:
INFO:Implementation:Building sikep503 - m4 - test
mupq/crypto_kem/test.c: In function 'write_canary':
mupq/crypto_kem/test.c:33:3: error: 'for' loop initial declarations are only allowed in C99 or C11 mode
   for (size_t i = 0; i < 8; i++) {
   ^
mupq/crypto_kem/test.c:33:3: note: use option -std=c99, -std=gnu99, -std=c11 or -std=gnu11 to compile your code
mupq/crypto_kem/test.c: In function 'check_canary':
mupq/crypto_kem/test.c:39:3: error: 'for' loop initial declarations are only allowed in C99 or C11 mode
   for (size_t i = 0; i < 8; i++) {
   ^
mupq/common/fips202.c: In function 'sha3_256_inc_finalize':
mupq/common/fips202.c:636:5: error: 'for' loop initial declarations are only allowed in C99 or C11 mode
     for (size_t i = 0; i < 32; i++) {
     ^
mupq/common/fips202.c:636:5: note: use option -std=c99, -std=gnu99, -std=c11 or -std=gnu11 to compile your code
mupq/common/fips202.c: In function 'sha3_384_inc_finalize':
mupq/common/fips202.c:684:5: error: 'for' loop initial declarations are only allowed in C99 or C11 mode
     for (size_t i = 0; i < 48; i++) {
     ^
mupq/common/fips202.c: In function 'sha3_384':
mupq/common/fips202.c:723:5: error: 'for' loop initial declarations are only allowed in C99 or C11 mode
     for (size_t i = 0; i < 48; i++) {
     ^
mupq/common/fips202.c: In function 'sha3_512_inc_finalize':
mupq/common/fips202.c:793:5: error: 'for' loop initial declarations are only allowed in C99 or C11 mode

yusimple, 2020-10-27:
I have include std=gnuc99, but it didn't work(I means another error occurs)

mkannwischer, 2020-10-27:
Seems like your compiler is defaulting to C89, which pqm4 is not compatible with. Adding `-std=gnu99` to your `CFLAGS` should indeed fix that. 

What other errors do you get?  What version of `arm-none-eabi-gcc` are you using? 
I did see some errors on older compiler versions before.

yusimple, 2020-10-27:
gcc version 4.9.3 20150529 (prerelease) (15:4.9.3+svn231177-1)

yusimple, 2020-10-27:
In file included from crypto_kem/ntrulpr761/m4f/kem.c:17:0:
crypto_kem/ntrulpr761/m4f/kem.c: In function '__UMULL':
crypto_kem/ntrulpr761/m4f/cmsis.h:685:117: warning: 'llr.w32[0]' is used uninitialized in this function [-Wuninitialized]
   __ASM volatile ("umull %0, %1, %2, %3" : "=r" (llr.w32[0]), "=r" (llr.w32[1]): "r" (op1), "r" (op2) , "0" (llr.w32[0]), "1" (llr.w32[1]) : "cc");
                                                                                                                     ^
crypto_kem/ntrulpr761/m4f/cmsis.h:685:135: warning: 'llr.w32[1]' is used uninitialized in this function [-Wuninitialized]
   __ASM volatile ("umull %0, %1, %2, %3" : "=r" (llr.w32[0]), "=r" (llr.w32[1]): "r" (op1), "r" (op2) , "0" (llr.w32[0]), "1" (llr.w32[1]) : "cc");
                                                                                                                                       ^
crypto_kem/ntrulpr761/m4f/Short_fromlist_asm_ntrulpr761.S: Assembler messages:
crypto_kem/ntrulpr761/m4f/Short_fromlist_asm_ntrulpr761.S:123: Error: invalid constant (2f9) after fixup
crypto_kem/ntrulpr761/m4f/Encode_asm.S: Assembler messages:
crypto_kem/ntrulpr761/m4f/Encode_asm.S:13: Error: invalid constant (11ef) after fixup
crypto_kem/ntrulpr761/m4f/Encode_asm.S:154: Error: invalid constant (5fb) after fixup
crypto_kem/ntrulpr761/m4f/Encode_asm.S:206: Error: invalid constant (23c5) after fixup
crypto_kem/ntrulpr761/m4f/Rq_mult3_asm.S: Assembler messages:
crypto_kem/ntrulpr761/m4f/Rq_mult3_asm.S:31: Error: invalid constant (11ef) after fixup
crypto_kem/ntrulpr761/m4f/Decode_asm.S: Assembler messages:
crypto_kem/ntrulpr761/m4f/Decode_asm.S:17: Error: invalid constant (9a3c) after fixup
crypto_kem/ntrulpr761/m4f/Decode_asm.S:18: Error: invalid constant (e361) after fixup
crypto_kem/ntrulpr761/m4f/Decode_asm.S:76: Error: invalid constant (5e95) after fixup
crypto_kem/ntrulpr761/m4f/Decode_asm.S:77: Error: invalid constant (64e3) after fixup
crypto_kem/ntrulpr761/m4f/Decode_asm.S:155: Error: invalid constant (3479) after fixup
crypto_kem/ntrulpr761/m4f/Decode_asm.S:156: Error: invalid constant (e361) after fixup
crypto_kem/ntrulpr761/m4f/Decode_asm.S:217: Error: invalid constant (f1ba) after fixup
crypto_kem/ntrulpr761/m4f/Decode_asm.S:218: Error: invalid constant (3d0f) after fixup
crypto_kem/ntrulpr761/m4f/Decode_asm.S:288: Error: invalid constant (f8cc) after fixup
crypto_kem/ntrulpr761/m4f/Decode_asm.S:289: Error: invalid constant (42b9) after fixup
crypto_kem/ntrulpr761/m4f/Decode_asm.S:353: Error: invalid constant (d50e) after fixup
crypto_kem/ntrulpr761/m4f/Decode_asm.S:354: Error: invalid constant (b833) after fixup
crypto_kem/ntrulpr761/m4f/Decode_asm.S:409: Error: invalid constant (9724) after fixup
crypto_kem/ntrulpr761/m4f/Decode_asm.S:410: Error: invalid constant (7e91) after fixup
crypto_kem/ntrulpr761/m4f/Decode_asm.S:464: Error: invalid constant (f5c3) after fixup
crypto_kem/ntrulpr761/m4f/Decode_asm.S:465: Error: invalid constant (5c29) after fixup
crypto_kem/ntrulpr761/m4f/Decode_asm.S:538: Error: invalid constant (cccd) after fixup
crypto_kem/ntrulpr761/m4f/Decode_asm.S:611: Error: invalid constant (f8d8) after fixup
crypto_kem/ntrulpr761/m4f/Decode_asm.S:612: Error: invalid constant (630d) after fixup
crypto_kem/ntrulpr761/m4f/Decode_asm.S:678: Error: invalid constant (d532) after fixup
crypto_kem/ntrulpr761/m4f/Decode_asm.S:679: Error: invalid constant (3d33) after fixup
crypto_kem/ntrulpr761/m4f/Rq_redp.S: Assembler messages:

yusimple, 2020-10-27:
after I use ` -std=gun99`, It occurs like this.

mkannwischer, 2020-10-27:
Seems like the NTRUPrime implementation is incompatible with older versions of `arm-none-eabi-gcc`. 
I've just pushed a fix to master. 

Can you please pull and check if it works now?

I've tested it with the most ancient version of `arm-none-eabi-gcc` that I was able to find (`5.3.1`). 
You may want to update at some point. It's 2020, already ;-).

yusimple, 2020-10-28:
this issues had been solved.

closed 2020-10-28 by yusimple