mupq/pqm4 #166: something wrong with `python3 test.py`

yusimple, 2020-10-28:
when I use `python3 test.py`,It occurs like below:
 INFO:Implementation:Building lightsaber - m4 - test
make: 'bin/crypto_kem_lightsaber_m4_test.bin' is up to date.
INFO:platform interface:Flashing bin/crypto_kem_lightsaber_m4_test.bin to device
WARNING:platform interface:timed out while waiting for the markers, reflashing
INFO:platform interface:Flashing bin/crypto_kem_lightsaber_m4_test.bin to device
Traceback (most recent call last):
  File "test.py", line 10, in <module>
    test.test_all(sys.argv[1:])
  File "/home/yuleitao/Documents/pqm4/mupq/mupq.py", line 248, in test_all
    if self.run_test(implementation) == -1:
  File "/home/yuleitao/Documents/pqm4/mupq/mupq.py", line 256, in run_test
    output = super().run_test(*args, **kwargs).strip()
  File "/home/yuleitao/Documents/pqm4/mupq/mupq.py", line 239, in run_test
    return self.interface.run(binary)
  File "/home/yuleitao/Documents/pqm4/mupq/mupq.py", line 169, in run
    self.flash(binary_path)
  File "/home/yuleitao/Documents/pqm4/interface.py", line 91, in flash
    stderr=subprocess.DEVNULL
  File "/usr/lib/python3.6/subprocess.py", line 311, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['st-flash', '--reset', 'write', 'bin/crypto_kem_lightsaber_m4_test.bin', '0x8000000']' returned non-zero exit status 255.

yusimple, 2020-10-28:
and my board is STM32F407G-DISC1,
my computer is ubuntu 1804

mkannwischer, 2020-10-28:
Can you please try running st-flash directly so that we can see what is going wrong?  
Assuming `bin/crypto_kem_lightsaber_m4_test.bin` already got assembled, run
```
st-flash write bin/crypto_kem_lightsaber_m4_test.bin 0x8000000
```
and paste the error here.

yusimple, 2020-10-28:
~/Documents/pqm4/bin$ st-flash write bin/crypto_kem_lightsaber_m4_test.bin 0x8000000
st-flash 1.6.1-109-g781557d
2020-10-28T02:30:17 INFO common.c: F4xx: 192 KiB SRAM, 1024 KiB flash in at least 16 KiB pages.
open(bin/crypto_kem_lightsaber_m4_test.bin) == -1
2020-10-28T02:30:17 ERROR common.c: map_file() == -1
stlink_fwrite_flash() == -1

yusimple, 2020-10-28:
but I just can't use `python3 test.py`

mkannwischer, 2020-10-28:
So flashing seems to be working fine. 
Can you get serial output by running 
``` 
python3 hostside/host_unidirectional.py
```

or 

```
screen /dev/ttyUSB0 115200
```

?

yusimple, 2020-10-28:
no, It just nothing

mkannwischer, 2020-10-28:
Then that's the problem. You cannot receive output from the board, so the scripts will keep retrying indefinitely. 
Can you make sure that your serial connection is working fine by flashing some other binaries, e.g., from https://github.com/joostrijneveld/STM32-getting-started?

Also please double check that your setup looks something like this and your RX pin is connected to PA2 on the board: 

[image]

yusimple, 2020-10-28:
did you mean mini B is for read, and PA2 for write?

mkannwischer, 2020-10-28:
PA2 is used for sending serial output to the host. 

The USB cable is used for flashing the binary to the board using st-link.

closed 2020-11-01 by yusimple