mupq/pqm4 #168: Repository package for serial module

GeorgeTasop, 2020-11-20:
Sorry if opening an issue is an overkill but I wanted somehow to inform you about it.
I am running Ubuntu 20.4 and as I was installing pqm4 i found out that the there is no apt package named `python-serial` but there is in fact one named `python3-serial` that just does it. That's it, have a nice day!

Ko-, 2020-11-20:
Hi! Some context: on Debian and Ubuntu, `python-serial` is the package for Python 2, and `python3-serial` is the package for Python 3. Python 2 has been unsupported and marked end-of-life since January 2020, so starting with Ubuntu 20.04, Ubuntu no longer ships with Python 2. For `pqm4`, it does say that the benchmarking scripts require Python >= 3.6 and in the instructions `python3` is called explicitly. But you're right, that paragraph about installing `pyserial` is not fully clear. What might have caused the confusion is that some of us use Arch Linux and there the packages are called `python2-pyserial` and `python-pyserial`, respectively, where `python` has been referring to Python 3 for years. In any case, the [upstream pyserial code base](https://github.com/pyserial/pyserial) actually supports both Python 2.7 and >= 3.4 simultaneously. I'll rephrase that paragraph a bit.

Ko-, 2020-11-20:
I hope it's more clear now!

closed 2020-11-20 by Ko-

GeorgeTasop, 2020-11-21:
> Hi! Some context: on Debian and Ubuntu, `python-serial` is the package for Python 2, and `python3-serial` is the package for Python 3. Python 2 has been unsupported and marked end-of-life since [...]

Ohh now I see. Thanks for the clarification. Cheers!

PR #169 (2020-11-20): Clarify pyserial installation instructions
Fixes #168
commit 34d6ed00 (2020-11-20): Clarify pyserial installation instructions (#169)

Fixes #168