mupq/pqm4 #204: Cannot get serial output

kzvkva, 2021-08-02:
Hi, when running `python3 hostside/host_unidirectional.py` I get only `> Returned data:` and no other output. Also, when I run `python3 test.py saber` I get only: 

> INFO:Implementation:Building saber - m4f - test
  CC      elf/crypto_kem_saber_m4f_test.elf
  OBJCOPY bin/crypto_kem_saber_m4f_test.bin
DEBUG:platform interface:Found start pattern: b'=========================\n'
INFO:BoardTestCase:Success
INFO:Implementation:Building saber - clean - test
  CC      elf/mupq_pqclean_crypto_kem_saber_clean_test.elf
  OBJCOPY bin/mupq_pqclean_crypto_kem_saber_clean_test.bin
DEBUG:platform interface:Found start pattern: b'=========================\n'
INFO:BoardTestCase:Success

When running 'python3 testvectors.py saber', it doesn't seem to work correctly too: 

> python3 testvectors.py saber
INFO:BoardTestCase:Running saber - clean on host
  HOST-CC obj-host/mupq/common/fips202.c.o
  HOST-CC obj-host/mupq/common/sp800-185.c.o
  HOST-CC obj-host/mupq/common/nistseedexpander.c.o
  HOST-CC obj-host/mupq/common/keccakf1600.c.o
  HOST-CC obj-host/mupq/pqclean/common/aes.c.o
  HOST-CC obj-host/mupq/pqclean/common/sha2.c.o
  HOST-AR obj-host/libsymcrypto.a
  HOST-LD bin-host/mupq_pqclean_crypto_kem_saber_clean_testvectors
INFO:Implementation:Building saber - m4f - testvectors
  CC      elf/crypto_kem_saber_m4f_testvectors.elf
  OBJCOPY bin/crypto_kem_saber_m4f_testvectors.bin
DEBUG:platform interface:Found start pattern: b'=========================\n'
INFO:Implementation:Building saber - clean - testvectors
  CC      elf/mupq_pqclean_crypto_kem_saber_clean_testvectors.elf
  OBJCOPY bin/mupq_pqclean_crypto_kem_saber_clean_testvectors.bin
DEBUG:platform interface:Found start pattern: b'=========================\n'

Is there any way to fix this? I am using STM32F407G-DISC1 board and  connected the pins like described (RX to PA2). 

Thanks!

mkannwischer, 2021-08-02:
This is intended behaviour and it all seems to be working fine. To get any output from `python3 hostside/host_unidirectional.py`, you will need to additionally flash a binary and run it. 
Try the following:
```
python3 test.py saber
python3 hostside/host_unidirectional.py
```
Then press the black reset button on your board. You should then get the serial output of the program that was last flashed. In this case `bin/mupq_pqclean_crypto_kem_saber_clean_test.bin`. 

You can also make other binaries and flash them manually, e.g., 
```
make -j4 PLATFORM=stm32f4discovery
st-flash write bin/mupq_pqclean_crypto_kem_kyber768_clean_test.bin 0x8000000
```

mkannwischer, 2021-08-03:
@kzvkva can you confirm that that works?

kzvkva, 2021-08-03:
thanks! it works now

closed 2021-08-03 by kzvkva