mupq/pqm4 #220: Rainbow Classic on STM32F407

DanielMilincic, 2021-12-12:
Hi, 
i am trying to benchmark rainbow classic on my stm32F407. I far as I understood rainbow1-classic shouldn't have too much memory consumption that it would not be feasible on a F407. 
But when I try to make the project with make -j4 PLATFORM=stm32f4discovery I get the error: 
`/Applications/ARM/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: elf/crypto_sign_rainbowI-classic_m4f_stack.elf section `.bss' will not fit in region `ram'
/Applications/ARM/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: region `ram' overflowed by 150760 bytes
collect2: error: ld returned 1 exit status
make: *** [elf/crypto_sign_rainbowI-classic_m4f_stack.elf] Error 1`

So it seems that it is not possible for me to use Rainbow on the STM32F407 after all?

Glad to receive any help, 
Daniel.

rpls, 2021-12-12:
Yeah, the implementation itself should have a memory consumption of about 445440 bytes, so no chance.You'll have to try the `nucleo-l4r5zi` platform.

DanielMilincic, 2021-12-12:
Oh I thought I read in the "Rainbow on Cortex-M4" that the size would be fairly small. But thank you for But thank you for pointing out!

closed 2021-12-12 by DanielMilincic