mupq/pqm4 #226: Questions about the range analysis of iNTT for "Faster Kyber and Dilithium on the Cortex-M4"

JunhaoHuang, 2022-03-03:
Hi team, I am reading the code regarding the recent paper "Faster Kyber and Dilithium on the Cortex-M4", and I have a question about the matrix-vector product and Better Accumulation part regarding the _f_stack_ version code. 

I see that using the better accumulation technique in the f_speed version code, we can reduce each element of the output vector down to (-q,q). Since poly_invntt is normally used after the matrix-vector product, the range of the input vector of **poly_invntt** lies in (-q,q) in the _f_speed_ version code. 

What I wonder is that in the _f_stack_ version code, the **matacc** function actually uses the previous double basemul accumulation function, and it should produce the result vector with element in (-kq, kq), k is the security parameter of Kyber. For Kyber1024, the range of each polynomial element that **invntt** takes should be (-4q,4q). However, the **invntt** function is the same as the f_speed version code. The first four layers of the light butterflies in **invntt** involve some additions and subtractions without multiplication. Therefore, For Kyber1024 in the _f_stack_ version code, two layers of addition/subtraction might overflow the int16_t. I wonder how you deal with this problem in f_stack code and why does it still work?

mkannwischer, 2022-03-03:
Thank you for your question. @dop-amin could please answer this one?

dop-amin, 2022-03-04:
Dear @JunhaoHuang,
thank you for your comment on the implementation. 
I looked into what you mentioned and I think that you are right. The implementation probably requires adaption to this inside the invntt (or at some point before). I will look into fixing this later this month. 
On your question on why it still works, I believe that this is an edge case that does not get triggered by the testing scripts.

Kind regards
Amin

mkannwischer, 2022-03-07:
@cryptojedi, @boyin, this may be an interesting possible bug for your formal verification projects.

mkannwischer, 2022-05-10:
This is resolved by #234.

mkannwischer, 2022-09-28:
Something seems odd. 
#234 was supposed to add an extra reduction to fastinvntt.S for the stack versions just like https://github.com/FasterKyberDilithiumM4/FasterKyberDilithiumM4/commit/daafc401da7d94e983552cbc3ffbde7df9f70c9d. 
However, that is not what #234 implements -- there is no change in the fastinvntt.S at all. 
I suspect that something went wrong when copying it over to pqm4. 

Please have a look @dop-amin. 

Thanks @marco-palumbi and @JunhaoHuang for pointing this out!

mkannwischer, 2022-10-17:
Hopefully finally addressed by https://github.com/mupq/pqm4/pull/242
Thanks @JunhaoHuang and @dop-amin!

closed 2022-10-17 by mkannwischer

marco-palumbi, 2022-10-17:
thanks to all!