mupq/pqm4 #228: make doesn't work properly for new folders

vincentvbh, 2022-03-16:
If one adds a new implementation by creating a folder under
```
crypto_kem/{scheme}/{build}
```
The python script works, but the makefile doesn't.
I tried
```
make PLATFORM=stm32f4discovery bin/crypto_kem_{scheme}_{build}_test.bin 
```
but it doesn't.
Please update the documentation on how to support new builds with only makefiles.

rpls, 2022-03-18:
That actually isn't supposed to happen, I'll look into it. Rough explanation for the mechanism:

1. The [`mupq/mk/schemes.mk` script](https://github.com/mupq/mupq/blob/0235e325b60dcbd41bc82cdee55b5e3940e70b3e/mk/schemes.mk#L26-L33) creates a list of implementations using `find`. So this is a "Makefile creates a Makefile" approach, and `make` _should_ rebuild this list every time its run (the target is declared `.PHONY`). Something clearly went wrong, which version of `make` to you use?
2. The python scripts also create a list of implementations, and if the python script calls `make` it explicitly names the implementation via the ` IMPLEMENTATION_PATH` variable. And if this variable is set, the mechanism in point 1 is skipped (because it does take a short moment, and the python scripts call `make` a lot).

vincentvbh, 2022-03-18:
I'm using ```GNU Make 3.81``` built for ```i386-apple-darwin11.3.0```.
Now I don't think this is a problem with new folders. Below is how I reproduced the problem:
``` cp -r crypto_kem/{scheme}/{build}``` to a place outside pqm4, ```rm -rf  crypto_kem/{scheme}/{build}```, and create a soft link to the folder outside pqm4.
I encountered an error saying that no rule for the target even if ```{scheme}``` and ```{build}``` are implementations already in pqm4.

@mkannwischer was not able to reproduce the problem, so I'm now throwing my files into pqm4 instead of creating soft links.

rpls, 2022-03-18:
Now that you say that you use a symlinked directory, I think an issue with `find`. The script only looks for directories (the `-type d` flag in [`schemes.mk`](https://github.com/mupq/mupq/blob/0235e325b60dcbd41bc82cdee55b5e3940e70b3e/mk/schemes.mk#L26-L33)). Right now, we only work with symlinked files. Could be, that the GNU `find` on most Linux distros accepts symlinks to directories, and the BSD `find` on macOS does not (by default).

vincentvbh, 2022-03-18:
I can confirm that adding ```-L``` right after ```find```  solves the issue.
I think I can close this issue now.
Thanks.

rpls, 2022-03-18:
Yeah, I'll close this here and probably work on a patch for the build system in `mupq` to make this more robust. While I don't see us ever using symlinks for implementation folders within the project, it's a useful feature to use `pqm4` as a vehicle to benchmark their own implementations in a somewhat standardized manner.

closed 2022-03-18 by rpls