mupq/pqm4 #229: Kyber512 shared secret is not correct when optimizations are enabled.

anhu, 2022-03-17:
When anything other than `-O0` is specified during compilation, the generated shared secret is different from the one generated by when it is using `-O0` .  Another way to state this is PQM4 compiled with `-O0` loses interoperability with PQM4 compiled with other optimization levels.  

Note: this was observed when compiling with STM32CubeIDE.

If you would like more details, please let me know.

mkannwischer, 2022-03-18:
Thank you for reporting this!
My best guess is that this is due to floating point registers not being saved. 
`s16-s31` are callee-save, so one should `vpush` them in the beginning (and then restore them in the end using `vpop`)  in case one wants to use them.
The current implementation does not do that, e.g., https://github.com/mupq/pqm4/blob/master/crypto_kem/kyber768/m4fspeed/fastntt.S or https://github.com/mupq/pqm4/blob/master/crypto_kem/kyber768/m4fspeed/matacc.S. 
In our test, this never showed up since our compiler did not make use of floating point registers. 

If you want to try, you can add the `vpush`/`vpop`.
We'll be preparing a patch soon. 

If you just want a working copy, I suggest you try the previous version: https://github.com/mupq/pqm4/tree/c37e5417322429230aa7992822e2a295e36133ea

vincentvbh, 2022-03-18:
We might also want to change this for dilithium3 since 769 is used

anhu, 2022-03-18:
Hello,

Thank you both for your prompt replies.  I would love to try your patch once it is done. For now, I have turned off optimizations when building for release mode in the IDE which makes everything work fine. Please update this ticket once the patch is complete.

Just thought I'd mention that this issue was found when doing an integration between pqm4 and wolfSSL on STM32 to enable post-quantum KEMs for key exchange in TLS 1.3. If you would like more information about it, please see the following blog post: 

https://www.wolfssl.com/post-quantum-tls-1-3-key-establishment-comes-stm32-cortex-m4/

Warm regards, Anthony

mkannwischer, 2022-05-10:
This should be resolved by #234. @anhu, can you please double check?

anhu, 2022-05-10:
Hello @mkannwischer , Unfortunately I am away from my lab on business travel right now. This will have to wait until next week.  I'll report here when I've tested against the new changes.

anhu, 2022-05-25:
Hello @mkannwischer , I was able to try your updates and unfortunately, it did not fix the issue. That is to say, I still need to use `-O0` .

anhu, 2022-05-25:
Please let me know if there any thing I can provide to you to better let you debug this issue.

rugo, 2022-06-03:
Hi @anhu, 

I'm gonna take a long shot here. Not sure if it will help, but I have a suspicion.

Is it possible that you have the path `PQClean/common` in your header search path? PQClean is a dependency of mupq, so it gets pulled when you clone PQM4 with submodules.

The reasoning behind this is that Kyber uses SHAKE internally. Now there are two `fips202.h` header. One in [PQClean](https://github.com/PQClean/PQClean/blob/master/common/fips202.h) and one in [mupq](https://github.com/mupq/mupq/blob/0235e325b60dcbd41bc82cdee55b5e3940e70b3e/common/fips202.h).

I can imagine that the following happens: Your build uses the header from PQClean.

PQClean defines the shake state as:

```c
typedef struct {
    uint64_t* ctx;
} shake128incctx;
```

instead of mupq, which defines the state as:

```c
typedef struct {
    uint64_t ctx[26];
} shake128incctx;
```

As you can see, the PQClean state is callee allocated. So what I could imagine happening, is that the Kyber code always writes out of the SHAKE state because it assumes the SHAKE state structure from mupq. However, when you don't use optimizations, there might be enough space on stack  so that nothing bad happens. Once you turn on optimizations, the space (coincidentally) available on the stack might be fewer and you overwrite things that should not be overwritten.

As I said, this is a complete long shot as I don't have access to the build or your environment. But maybe check if `PQClean/common` is in the search path before `mupq/common`.

anhu, 2022-06-03:
Hi @rugo !! 

Thank you for letting me know about your very reasonable long-shot suspicion!

Unfortunately, what you describe is not what I'm seeing.  I have the following in my fips202.h: 
```
// Context for incremental API
typedef struct {
    uint64_t ctx[26];
} shake128incctx;
```
So I am using the one from mupq. 

> I don't have access to the build or your environment. 

If you would like to reproduce my build and environment, you can have a look at our documentation which describes how to setup everything: https://github.com/wolfSSL/wolfssl-examples/tree/master/pq/stm32

It is quite an involved process as an IDE is involved so there is a youtube video link there that describes part of the setup. 

Warm regards, Anthony

mkannwischer, 2022-09-26:
Dear @anhu, 
Can you please try again after #240 got merged. That may well explain your issues. 
Please let me know if that helps.

Cheers, 
Matthias

anhu, 2022-09-26:
Hello @mkannwischer, 

Thanks for merging that in!  I  will take it for a try as soon as possible, however I've just returned from an extended period of travel and several other items need to be resolved.  Please expect a delay in my attempt. 

Warm regards, Anthony

anhu, 2022-11-24:
@mkannwischer! 

Success!  I can turn on all levels of optimizations!! This issue is now resolved! :+1: @dop-amin , thank you very much for your fix. 

Please accept my apologies for the delay in this confirmation. 

I will now go ahead and close this issue. 

Warm regards, Anthony

closed 2022-11-24 by anhu