mupq/pqm4 #239: Doubt on the SPHINCS+ implementation

samyuktha123sets, 2022-07-14:
Hi Team,

If I want to run SPHINCS+ from the PQM4 on my Cortex M4 board, will it run? As M4 optimised code is not available for SPHINCS+ and is only available in the PQClean .  Does that mean PQClean implementation of SPHINCS+ will run on Cortex M4 ?

mkannwischer, 2022-07-14:
We run the SPHINCS+ implementation with optimized code for Keccak. 
See https://github.com/mupq/pqm4/blob/master/common/keccakf1600.S
It should be fairly okay. 

The SHA2 implementation is here: https://github.com/mupq/mupq/blob/0235e325b60dcbd41bc82cdee55b5e3940e70b3e/common/sha2.c 
It should also not be too bad.

samyuktha123sets, 2022-08-27:
Hi,
Is the keccak code referred here a constant time implementation?

mkannwischer, 2022-08-27:
yes

closed 2022-09-26 by mkannwischer