mupq/pqm4 #243: Stack benchmark not working for falcon-512

rolandjeg, 2022-10-14:
I try to run the stack usage benchmark for the falcon-512 signature scheme and it does not return any output. Falcon-1024 on the other hand runs fine and returns the following:

`> Returned data:
==========================
keypair stack usage:
1256
sign stack usage:
2544
verify stack usage:
412
Signature valid!

#`

The Board I use is the stm32f407G Discovery board.

mkannwischer, 2022-10-17:
I have tried this on my machine and all 3 implementations of falcon-512 work: 
```
==========================
keypair stack usage:
1344
sign stack usage:
2556
verify stack usage:
412
Signature valid!

#
==========================
keypair stack usage:
1344
sign stack usage:
2556
verify stack usage:
412
Signature valid!

#
==========================
keypair stack usage:
1296
sign stack usage:
2620
verify stack usage:
412
Signature valid!

#
```

Could you please provide some more information on the compiler version you are using and where you got it from? 

I'm using `arm-none-eabi-gcc (Arm GNU Toolchain 11.3.Rel1) 11.3.1 20220712` from https://developer.arm.com/downloads/-/arm-gnu-toolchain-downloads

rolandjeg, 2022-10-17:
That was indeed the culprit. I used version 12.1.0 from the Arch repository, now that I use 11.3.Rel1 everything works as intended.

More information about the version where it didn't work:
```
❯ arm-none-eabi-gcc -v
Using built-in specs.
COLLECT_GCC=arm-none-eabi-gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/arm-none-eabi/12.1.0/lto-wrapper
Target: arm-none-eabi
Configured with: /build/arm-none-eabi-gcc/src/gcc-12.1.0/configure --target=arm-none-eabi --prefix=/usr --with-sysroot=/usr/arm-none-eabi --with-native-system-header-dir=/include --libexecdir=/usr/lib --enable-languages=c,c++ --enable-plugins --disable-decimal-float --disable-libffi --disable-libgomp --disable-libmudflap --disable-libquadmath --disable-libssp --disable-libstdcxx-pch --disable-nls --disable-shared --disable-threads --disable-tls --with-gnu-as --with-gnu-ld --with-system-zlib --with-newlib --with-headers=/usr/arm-none-eabi/include --with-python-dir=share/gcc-arm-none-eabi --with-gmp --with-mpfr --with-mpc --with-isl --with-libelf --enable-gnu-indirect-function --with-host-libstdcxx='-static-libgcc -Wl,-Bstatic,-lstdc++,-Bdynamic -lm' --with-pkgversion='Arch Repository' --with-bugurl=https://bugs.archlinux.org/ --with-multilib-list=rmprofile
Thread model: single
Supported LTO compression algorithms: zlib zstd
gcc version 12.1.0 (Arch Repository)
```

mkannwischer, 2022-10-18:
Thanks! I ran into many similar issues with previous versions from the Arch repo and gave up fixing our code.
I'm now just using the most recent one from Arm. 

Can I close this issue or do you want to investigate further what is causing the problem?

closed 2022-10-18 by mkannwischer