mupq/pqm4 #245: Is the "mk/ cw308t-stm32f3.mk" is imcomplete?

lkcp, 2022-10-19:
Hi,

When I planned to compiled **pqm4** for cw308t-stm32f3 platform, I met several problems such as "memory overflow" or something strange. I found many chioces in **pqm4** is not suitable for cw308t-stm32f3, such as ntru*, dilithium*, but they are not included in the "mk/ cw308t-stm32f3.mk", so I 'm not sure if the errors is as expected. 

To my best knowledge, I thought it is correct to add more choices into "mk/ cw308t-stm32f3.mk". For example, in dilithium's sign, a stack space exceeds **40KB**(RAM in stm32f3) is needed, that means dilithium shouldn't be implemented in cw308t-stm32f3 platform.

Best regards, Jian

mkannwischer, 2022-10-21:
Dear Jian, 

The `cw308t-stm32f` should be working, yes. 
Regarding the RAM issue, pqm4 should be automatically skipping implementations that require too much memory. pqm4 currently lists `cw308t-stm32f3` with 64 KiB, so I think that needs to change to 40 KiB. I'll change that.

Regarding NTRU, you are probably also getting: 
```
region `rom' overflowed by 18892 bytes collect2: error: ld returned 1 exit status
```
i.e., the code of the implementation is larger than the available flash (256 KiB). Not much we can do there. We could automatically skip those, but since NTRU is going to be deleted in #238 anyway, I think it is not needed unless other implementations are also affected.

mkannwischer, 2022-10-21:
Actually, I'm fine with manually adding those to `mk/cw308t-stm32f3.mk` while we wait for #238 to be merged.

lkcp, 2022-10-21:
Dear Matthias,

Thany you for your reply, that helps a lot.

Best regards, Jian

mkannwischer, 2022-10-22:
Let's keep this open until we have actually done something about it.