mupq/pqm4 #246: build error

derstrand, 2022-10-21:


derstrand, 2022-10-21:
Dear pqm4 community,

I have tried to build pqm4 library on an Ubuntu host. Followed all steps outlined, and tried to build for mps2-an386 platform. Build fails with the following errors. 

[image]

Best,
@derstrand

mkannwischer, 2022-10-22:
Do you mind trying a newer compiler from https://developer.arm.com/downloads/-/arm-gnu-toolchain-downloads and/or let me know what compiler you are using

derstrand, 2022-10-24:
Thank you, @mkannwischer. I will have a look at the those compilers. Currently, I have:
derstrand> gcc --version
gcc (Ubuntu 7.5.0-3ubuntu1~18.04) 7.5.0

mkannwischer, 2022-12-10:
I am assuming this resolved the issue? 
Closing this.

closed 2022-12-10 by mkannwischer