mupq/pqm4 #248: make failure

yberlin, 2022-11-19:
I clone the latest version of pqm4, and tried to make like before:

make IMPLEMENTATION_PATH=crypto_kem/kyber768/m4fspeed PLATFORM=stm32f4discovery

After some related files compiled, here comes a error:

  AR      libopencm3_pac55xx.a
make[1]: Leaving directory '/home/berlin/Documents/20221119_test/pqm4/libopencm3'
make: *** No rule to make target 'mupq/common/nistseedexpander.c', needed by 'elf/crypto_kem_kyber768_m4fspeed_test.elf'.  Stop.

The environment is a VMWare workstation of ubuntu. And older versions of pqm4 can be made successfully.
Could you please help me to fix this?

yberlin, 2022-11-20:
I tried to figure out the problem. And I found that the file nistseedexpander.c and .h and randombytes.h in mupq/common are broken.
In ubuntu, the properties of the file show that the type is :Link (broken) (inode/symlink)
Even though I copy the file from mupq repo, the file will not work. Do you know why this occurs?

mkannwischer, 2022-11-22:
I think what you are seeing is the result of not recursively cloning the dependencies.
Please make sure to either 
```
git clone --recurse-submodules https://github.com/mupq/pqm4
```

or 
```
git clone https://github.com/mupq/pqm4
git submodule update --init --recursive
```

I think what you did (correct me if I am wrong) was 
```
git clone https://github.com/mupq/pqm4
git submodule update --init
```
That results in the [pqclean](https://github.com/PQClean/PQClean) dependency of [mupq](https://github.com/mupq/mupq) to not exist. 

Please let me know if that solves your problem.

yberlin, 2022-11-29:
Sorry for the delay.
And thank you for your reply!
Actually, I fixed the problem by cloning the mupq repo separately and copying this repo into pqm4.
Your method may also work. I really used "recurse" command to clone before. But maybe the repo is not completely cloned because the problem of network.

closed 2022-11-29 by yberlin