mupq/pqm4 #249: Missing Output directories including *.a, *.bin etc. after Build

Nouri-Alnahawi, 2022-11-24:
Dear pqm4 team,

I've followed the instructions of building the pqm4 library for the nucleo-l4r5zi platform as explained in the README.md. However, I cannot seem to find and locate the output directories and files as described in the README.md of the mupq sub-repo.

It says that library files, executables, as well as assembler executables are generated for each included scheme, in addition to the related object files. All I'm getting after a make run are the .bin test files in the level one directory /bin.

Am I missing something here? We want to link api files from the optimized schemes (e.g. Kyber & Dilithium) into our own application, but we've kind of running in circles for a while now :D 

Thanks in advance!

Cheers
Nouri

mkannwischer, 2022-11-24:
Dear Nouri, 
Thanks for your interest in pqm4. 
The pqm4 build system is not meant to build a library that can be linked into another project. It is merely building binaries for testing and benchmarking using the python framework. 

If you want use it in another application, you can simple copy the files from the respective scheme folders in the `crypto_kem` and `crypto_sign` folders. Additionally, you will need some for generating random numbers (randombytes.c) and the the common symmetric code algorithm (SHA-3, SHA-2, or AES depending on scheme and parameter set). It should all be fairly straightforward as each implementation has all files in one folder just as is SUPERCOP. 

Let me know if that helps.

Cheers, 
Matthias

mkannwischer, 2022-12-10:
Closing this as I have not heard anything back. Please let me know if there is still anything unclear.

closed 2022-12-10 by mkannwischer