mupq/pqm4 #251: BIKE implementation vulnerable/out of date

ambiso, 2023-02-23:
Hi,

the current BIKE implementation uses rejection sampling which is non-constant time in the used seed. The seed is derived from the message obtained from decrypting the ciphertext, therefore the seed is to be treated as a secret.
This permits a key-recovery timing side-channel attack; see [1,2].

The BIKE spec has been updated to use (a slightly modified version of) the countermeasure described in [3]. This new sampling method (most likely) generates different vectors under for the same message, so implementations using different sampling methods are also incompatible.

If the BIKE implementation in this repository is meant to be constant-time/timing side-channel free, probably this is one of the things that should be addressed. Alternatively, a notice that the implementation is incompatible with the latest BIKE spec should be created.

Kind regards,
ambiso

[1]: https://eprint.iacr.org/2021/1485
[2]: https://eprint.iacr.org/2021/1631.pdf
[3]: https://bikesuite.org/files/v5.0/BIKE_Spec.2022.10.04.1.pdf

37eex9, 2023-10-26:
Hey,
I figured out the same thing and started fixing it. I recently made a pull request for mupq/mupq#117 and when this is accepted I will make one for pqm4, containing the latest changes in mupq.

Within the last weeks, there was a lot of activity on this repos, hopefully the prs will be accepted soon.

Kind regards,
Till

37eex9, 2023-11-26:
With the merge of #303, the implementations were updated. So this issue can be closed.

closed 2023-11-26 by rpls