mupq/pqm4 #252: Problem of running test.py

D0641826, 2023-03-01:
Hi, I am new to this project.

I'm using cw308t-stm32F3 as my platform and I'm going to run kyber768, when I run python3 test.py, I met the problem called "raise Exception(f"Unexpected path format:'{path}'")
Exception: Unexpected path format:'crypto_kem\kyber768\m4fspeed'". in the mupq.py.
Which configuration does I need to fix, or how can I solve this problem. 

Best regards, Ricky

rpls, 2023-03-01:
Hey,

could you give me the full command line you used to start `test.py`?

Cheers, Richard

rickyhsyk, 2023-03-02:
Dear Richard,
I'm using the Chipwhisperer Lite together with the CW308 UFO with the STM32F3 Target and I'm running it on  win10.

$ python3 test.py kyber768
Traceback (most recent call last):
  File "C:\Users\admin\ChipWhisperer5_64\cw\pqm4\test.py", line 12, in <module>
    test.test_all(rest)
  File "C:\Users\admin\ChipWhisperer5_64\cw\pqm4\mupq\mupq.py", line 197, in test_all
    for implementation in self.get_implementations():
  File "C:\Users\admin\ChipWhisperer5_64\cw\pqm4\mupq\mupq.py", line 128, in get_implementations
    impl = Implementation.from_path(parent, path, namespace, self.makeflags)
  File "C:\Users\admin\ChipWhisperer5_64\cw\pqm4\mupq\mupq.py", line 53, in from_path
    raise Exception(f"Unexpected path format:'{path}'")
Exception: Unexpected path format:'crypto_kem\kyber768\m4fspeed'

Before running this, I install the chipwhisperer library and ran the example code on it to make sure it can connect successfully.
Best regards, Ricky

rpls, 2023-03-03:
Sorry, running on Windows isn't really supported, so you'll have to work on a Linux or macOS platform. Running on the "Windows Subsystem for Linux" should probably work, although I'm uncertain if one can pass USB devices through to the subsystem.

Furthermore, unless specified with a `-p` argument, all the scripts assume that the `stm32f4discovery` platform is used. So to run this on the ChipWhisperer, you'll have to supply the python scripts with the following flag: `-p cw308t-stm32f3`

rickyhsyk, 2023-03-06:
Okay, I'll try this on Linus platform to checkout if there's other question. Thank you for answering my question.

Best regards, Ricky

rickyhsyk, 2023-03-09:
Hi, sorry for borrowing,  I had change my platform to Linux, the previous problem had been solved but it got an new problem:

INFO:Implementation:Building kyber768 - m4fstack - test
Variable PLATFORM changed, forcing rebuild!
  GEN 	obj/.config.mk
/bin/sh: 1: cannot create obj/.config.mk: Permission denied
make: *** No rule to make target 'obj/.config.mk', needed by 'obj/generated.stm32f303rct7.ld'.  Stop.
Traceback (most recent call last):
  File "/home/cwuser777/Desktop/chipwhisperer/software/chipwhisperer/pqm4/test.py", line 12, in <module>
	test.test_all(rest)
  File "/home/cwuser777/Desktop/chipwhisperer/software/chipwhisperer/pqm4/mupq/mupq.py", line 202, in test_all
	if self.run_test(implementation) == -1:
  File "/home/cwuser777/Desktop/chipwhisperer/software/chipwhisperer/pqm4/mupq/mupq.py", line 210, in run_test
	output = super().run_test(*args, **kwargs).strip()
  File "/home/cwuser777/Desktop/chipwhisperer/software/chipwhisperer/pqm4/mupq/mupq.py", line 189, in run_test
	implementation.build_binary(f'{self.test_type}',
  File "/home/cwuser777/Desktop/chipwhisperer/software/chipwhisperer/pqm4/mupq/mupq.py", line 77, in build_binary
	self.run_make(self.get_binary_path(test_type, bin_type))
  File "/home/cwuser777/Desktop/chipwhisperer/software/chipwhisperer/pqm4/mupq/mupq.py", line 67, in run_make
	return subprocess.check_call(makeflags)
  File "/home/cwuser777/.pyenv/versions/3.9.5/lib/python3.9/subprocess.py", line 373, in check_call
	raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['make', 'IMPLEMENTATION_PATH=crypto_kem/kyber768/m4fstack', 'PLATFORM=cw308t-stm32f3', 'MUPQ_ITERATIONS=1', 'LTO=', 'AIO=1', 'bin/crypto_kem_kyber768_m4fstack_test.hex']' returned non-zero exit status 2.
I had ran the command and change the platform to chipwhisperer, but no idea which parameter I need to fix

Best regards, Ricky

closed 2023-03-13 by rickyhsyk