mupq/pqm4 #255: Question to Memory Benchmarks

Nouri-Alnahawi, 2023-07-13:
Dear Mr. Kannwischer,
dear pqm4 team,

I was trying to figure out how you collect the memory benchmarks within the tests, bu unfortunately I couldn't overcome the complexity of the source code. I understand that you use python scripts to get the data, but I cannot find where the actual memory benchmarks take place in the C sources.

We are trying to run similar benchmarks on an application using your PQC implementations  , and so far we were able to use DWT for the clock cycles, and the `arm-none-eabi-sizes` for static memory. However we are failing to obtain the dynamic memory allocations and the stack memory.

I would really appreciate an input on this matter. Thank you in advance!

Kind Regards
Nouri Alnahawi

P.S.: We met briefly in Toyko on RWC, but I don't think you would remember me :)

rpls, 2023-09-06:
Hey Nouri-Alnahawi,

sorry for the delay, only recently got the time for more project maintenance. If it's still relevant: The memory measurement in the benchmarks tries to measure the stack with "stackspraying". The project defines a hardware abstraction layer that offers a `hal_get_stack_size` function. This function returns how much space there is between the current stack pointer and the end of the heap.

https://github.com/mupq/pqm4/blob/a525417134995302bb5013dd112dec65cdb28ca9/common/hal-opencm3.c#L355 

The stack test simply calls this function, writes a data pattern from the end of the heap to the top of the stack (i.e., all currently unused memory), and then calls the cryptographic function to be measured. After the function is done, it checks how much of the pattern is still intact, starting from the top of the stack. The amount of disturbed data is used as a measurement of how much stack is used.

https://github.com/mupq/mupq/blob/2425e129957321f6ae629ac28dbe2c34bdf268ce/crypto_kem/stack.c#L45

It's somewhat crude, but works well enough. A better option would be some form of static analysis, but that's not possible for all implementations (and hard).

Hope this helps.

Cheers
Richard

closed 2023-09-06 by rpls

Nouri-Alnahawi, 2023-09-12:
Hello Richard,

no worries. Thank you very much for the answer!

Cheers
Nouri