mupq/pqm4 #258: support for platform cw308t-stm32f4

JLnukeNV, 2023-10-09:
Hello, I'm trying to test kyber768-m4f implementation on platform cw308t-stm32f4, stm32f415rgt6 target board to be exact. As we know, if we use cw308t-stm32f3, the randombytes will be implemented by software, the difference bewteen them is cw308t-stm32f4 should implement the randombytes using rng_get_random_blocking() in rng.h. But from my testing, the code stucked at randombytes for some reason(s) I don't know. Can you support cw308t-stm32f4 platform and support its rng feature.

rpls, 2023-10-09:
Are you sure about the Target board? The NewAE website lists the `STM32F405RGT6` as the chip.

JLnukeNV, 2023-10-10:
Yes I'm sure it's stm32f415rgt6, check https://rtfm.newae.com/Targets/UFO%20Targets/CW308T-STM32F/, you will find stm32f405rgt6 on mouser not available, so most will choose stm32f415rgt6(Hardware Crypto) instead, which have 21 stocks by the time I check it on mouser.

rpls, 2023-10-10:
I'll add them in the next few days, but I don't have the targets myself, so you'll have to test them. I'll open up a branch with the targets supported, when I'm ready.

JLnukeNV, 2023-10-11:
Glad to hear that. I 'll test them and report the result here.

rpls, 2023-10-22:
Sorry for the delay. Could you test the code in #259? I.e., check out the branch `opencm3optim`? Running the code expects an environment as present when the ChipWhisperer is connected. I.e., the target board expects the clock to be supplied from the CW. And it asumes the default CW settings, with a clock of 7372800Hz (the default value of CW last I checked). If you run the code as is with the CW308 Ufo board and a chipwhisperer connected, you shouldn't have any problems. If you want to run the UFO board standalone, you'll have to supply a clock somehow...

37eex9, 2023-10-24:
As I just realize, I'm late a few days.
Within the last weeks, I added support for `stm32f415` in my local repo, too. Thanks for already pushing your efforts. I'm afraid you do not implement the hardware rng support (or I missed it).  According to the [chipwhisperer hal](https://github.com/newaetech/chipwhisperer/blob/develop/hardware/victims/firmware/hal/stm32f4/stm32f4_hal.c) we need to enable `HSI` and `PLL` for hw rng.
So in my hal setup the following lines are present and do not occur in your setup:
```
const struct rcc_clock_scale cw_clock = {
  .pllm = 12,
  .plln = 196,
  .pllp = 4,
  .pllq = 7,
  .pllr = 0,
  .flash_config = FLASH_ACR_LATENCY_0WS,
};
```
```
  flash_set_ws(cw_clock.flash_config);
  rcc_set_sysclk_source(RCC_CFGR_SW_HSE);
  rcc_wait_for_sysclk_status(RCC_HSE);
  
  /* HSI and PLL needed for RNG
  that's why we can't use rcc_clock_setup_pll()*/
  rcc_osc_on(RCC_HSI);
  rcc_wait_for_osc_ready(RCC_HSI);
  rcc_osc_off(RCC_PLL);
  rcc_set_main_pll_hse(cw_clock.pllm, cw_clock.plln,
        cw_clock.pllp, cw_clock.pllq, cw_clock.pllr);
  
  rcc_osc_on(RCC_PLL);
  rcc_wait_for_osc_ready(RCC_PLL);

  rcc_periph_clock_enable(RCC_RNG);
  rng_enable();
```

Before creating a pull request, I wanted to discuss it here, if this makes sense.

But one thing which is missing is this entry ` 'cw308t-stm32f415': 192*1024,` in the dict `platform_memory`, line 65 in [interface.py](https://github.com/mupq/pqm4/blob/master/interface.py).

Nevertheless, thanks for your work :)

JLnukeNV, 2023-10-25:
@37eex9 
Excuse me, did that mean from your testing stm32f415 should work fine after setting the `platform_memory` correctly without hardware rng?

It seems I have a problem, not sure where it is, the hardware or the software. I tested the code by running`python test.py - p cw308t-stm32f415`, but my board stuck after stm32f programmer `Verified flash OK`. By the way, stm32f303 works just fine.

37eex9, 2023-10-26:
@JLnukeNV
Which scheme was flashed to your stm32f415? E.g., __bikel3__ needs too much memory for `decapsulation` and crashes the board, though it is supposed to fit. Even __bikel1__ is quite resource exhaustive and with the board only running at a 7.3MHz clock, __bikel1\_mf4\_test__ takes about 15min to complete.
`python test.py - p cw308t-stm32f415` seems to work with the `hal_setup` that I suggest. With yours I can confirm that it gets stuck.

JLnukeNV, 2023-10-27:
It stuck at the first scheme to be tested `kyber512-90s - m4fstack`, which should be done within 10 seconds at 7.3Mhz clock.

37eex9, 2023-10-27:
@JLnukeNV in my fork I implemented the changes as suggested above. You are right, for kyber512-90s the target takes a few seconds.
Can you verify, that the branch [stm32f415](https://github.com/37eex9/pqm4/tree/stm32f415) works for you as expected, too.

Sorry, I was confused and thought you implemented the clock in the first place.

JLnukeNV, 2023-11-05:
I think this issue is solved by both you and 37eex9, if you look at the solution 37eex9 had given. Should I close this issue? Or we need more discuss? @rpls

37eex9, 2023-11-10:
I would not yet close the issue, as the changes are not included in the master branch.
@rpls Do you want to change `common/hal-opencm3.c` or am I supposed to open a pull request? Do you need benchmarks to include a new platform?

rpls, 2023-11-12:
The `opencm3` platform code is becoming a mess, I think I'll do a bigger cleanup. But not sure when I'll get to it, so feel free to do a small PR. Also, there seem to be a couple of people in this thread using the chipwhisperer platform, so I have a couple of questions:
- Which Targetboards do you have, or which ones would you like to see support for?
- Are you happy with running things in default ChipWhisperer settings? Or do you maybe want more support for custom clocking (as in, not just nice to have, but actually need it)?

JLnukeNV, 2023-11-20:
I can't stand for all ChipWhisperer users, I'm just a guy who wants to reproduce papers.
For me, use targetboards that most paper use is a efficient way to reproduce. So I seeked cw308t-stm32f415.

Chipwhisperer Lite has limited buffer size, only 24400 points can be stored. If we don't want to change the sampling rate of the scope`(96MS/s)`, then we should run the board at a higher clock rate and the time per instuctions become lower. So we can capture power traces with more operations. Set the clock rate at 24Mhz may be helpful according to 'https://doi.org/10.46586/tches.v2021.i4.676-707'.

37eex9, 2023-11-26:
@JLnukeNV, if I get you correctly, the `decimate` setting should do the same trick. [documentation](https://chipwhisperer.readthedocs.io/en/latest/scope-api.html#chipwhisperer.capture.scopes._OpenADCInterface.TriggerSettings.decimate)
But maybe that's what you mean with changing the sampling rate.

@rpls, for me, the supported targets stm32f303 and stm32f415, seem to be a good choice. Because the former is well distributed with the CW-Lite ARM and the latter is best equipped with RAM and flash. When looking at the [supported devices](https://rtfm.newae.com/Targets/UFO%20Targets/CW308T-STM32F/#supported-devices) for the UFO, there is only stm32l562re target which offers more RAM than stm32f415, but this has the Cortex-M33 architecture. So, I assume it does not really belong into this repo.

From my point of view, an alternative clock rate is more of a nice to have, as it would speed up experiments.

closed 2023-12-29 by JLnukeNV