mupq/pqm4 #310: Issues with benchmarks on stm32f4discovery from commit 8582f47bba034524c87d8e1630553d799783a61d

marco-palumbi, 2023-11-14:
Starting from commit 8582f47bba034524c87d8e1630553d799783a61d benchmarks doesn't work for me anymore on stm32f4discovery.

for example if I run:

./benchmarks.py -p stm32f4discovery kyber512
the script last forever... indeed the board output only the start pattern "==========================" and no more characters.

It seems that the new measurement HAL API is not working.

before commit 8582f47bb all seems to work fine for me.

I am using compiler **arm-none-eabi-gcc (15:10.3-2021.07-4) 10.3.1 20210621 (release)** in **Ubuntu 22.04.2 LTS**

Which compiler are you using?

rpls, 2023-11-14:
I'll look into it. As for the compiler: we're using the GNU toolchain supplied by ARM, which right now is based on GCC 13.2.

https://developer.arm.com/downloads/-/arm-gnu-toolchain-downloads

rpls, 2023-11-14:
OK, so because we're currently switching to the Nucleo-L4R5ZI board, I didn't properly test the new stack measurement on the old boards. The linkerscripts aren't right for that. VERY quick fix is the following patch:

```diff
diff --git a/ldscripts/stm32f4discovery.ld b/ldscripts/stm32f4discovery.ld
index a3fb9f1..fb18aec 100644
--- a/ldscripts/stm32f4discovery.ld
+++ b/ldscripts/stm32f4discovery.ld
@@ -58,6 +58,7 @@ SECTIONS
   *(COMMON)
   . = ALIGN(4);
   _ebss = .;
+  end = .;
  } >ram
  .ram2 : {
   *(.ram2*)
@@ -69,6 +70,5 @@ SECTIONS
  } >ccm
  /DISCARD/ : { *(.eh_frame) }
  . = ALIGN(4);
- end = .;
 }
 PROVIDE(_stack = ORIGIN(ram) + LENGTH(ram));
```

For some reason, the error doesn't pop up on my MacOS machine, which should be an entirely different issue in itself... But apply the one above, it should work. I'll fix this in a separate test for all the other platforms...

marco-palumbi, 2023-11-14:
With the patch it works fine!
Thanks

marco-palumbi, 2023-11-16:
just to add info: dilithium5 still not working even with the patch...
was working @e21677c

mkannwischer, 2026-08-25:
Heads-up: pqm4 is being wound down and will be archived in or after January 2027, see #409. We are unlikely to address this ourselves, but pull requests are welcome until **30 November 2026**.

vincentvbh, 2026-10-18:
Fixed, see #416.

Drafted with Claude Opus 5.5.

closed 2026-10-18 by vincentvbh