mupq/pqm4 #326: build error using Keil

qinleiyong, 2024-01-19:
My Keil version is V5.33. When I try to compile the kyber748 project alone, a lot of errors appear：<instantiation>(12): error: invalid instruction  ：ubfx.w tmp, tmp4, #0, #12；<instantiation>(4): error: too many operands for instruction ldrh.w tmp2, [aptr], #6;...........................My compile options are "Assembler option = armclang(Auto select) ".

rpls, 2024-01-19:
I'm not familiar with the Keil toolchain, I guess they have a Clang based compiler? I think most assembly files assume a GCC/Binutiles based tool chain. Which MCU are you targeting?

ate-wang, 2025-01-22:
I want to know if anyone has tested the performance of Kyber and Dilithium in the M4 environment, and what can be achieved in the M4 environment? This includes key pair generation, encapsulation, decapsulation, signing, signature verification, etc. Has anyone done relevant tests that you can share with me?

mkannwischer, 2025-01-22:
> I want to know if anyone has tested the performance of Kyber and Dilithium in the M4 environment, and what can be achieved in the M4 environment? This includes key pair generation, encapsulation, [...]

All the information you are looking for can be found in https://github.com/mupq/pqm4/blob/master/benchmarks.md
I have received your question by e-mail less than 8 hours ago and there is no need to post the same on in multiple unrelated threads on Github.

mkannwischer, 2026-08-25:
Heads-up: pqm4 is being wound down and will be archived in or after January 2027, see #409. We are unlikely to address this ourselves, but pull requests are welcome until **30 November 2026**.

vincentvbh, 2026-10-18:
pqm4 targets the GNU toolchain (arm-none-eabi-gcc); its assembly is written
for the GNU assembler, and other toolchains such as Keil's are not
supported. For a maintained ML-KEM that caters to embedded use, see
mlkem-native, as the README notes. Closing.

Drafted with Claude Opus 5.5.

closed 2026-10-18 by vincentvbh