mupq/pqm4 #330: The result of kyber768 project is abnormal.

qinleiyong, 2024-01-23:
I tested the current master version and the Release-Round3 version of the kyber768 project. The crypto_kem_enc and crypto_kem_dec results are wrong, but the results of Release-Round2 are normal.

rpls, 2024-02-27:
We're currently updating all benchmarks to use the most up-to-date ARM toolchain (which changes a lot of figures) and also use a new board Nucleo-L4R5ZI. Should be done in time for the next NIST PQC workshop.

mkannwischer, 2026-08-25:
Heads-up: pqm4 is being wound down and will be archived in or after January 2027, see #409. We are unlikely to address this ourselves, but pull requests are welcome until **30 November 2026**.

vincentvbh, 2026-10-17:
Thanks for the report. Since then, Kyber has been replaced by ML-KEM (FIPS 203) in pqm4. I re-tested ML-KEM-512/768/1024 (clean, m4fspeed, m4fstack) on current master, both in QEMU (mps2-an386) and on a Nucleo-L4R5ZI: all tests and testvectors pass, and 500 key exchanges per ML-KEM-768 implementation gave no errors. I'm closing this as resolved; please reopen with your toolchain and board details if you still see wrong results.

closed 2026-10-17 by vincentvbh