mupq/pqm4 #334: OS problem

rickyhsyk, 2024-03-04:
Hi, I'm using cw308t-stm32f3 as by targetboard, I'm trying to connect with simpleserial on chipwhisperer, but the problem occurs when it goes to randombytes.c:

  /usr/lib/gcc/arm-none-eabi/10.3.1/../../../arm-none-eabi/bin/ld: objdir-CW308_STM32F3/pqm4/randombytes.o: in function `PQCLEAN_randombytes':
/home/cwuser777/Desktop/chipwhisperer/hardware/victims/firmware/simpleserial-pqm4/pqm4/randombytes.c:336: undefined reference to `randombytes_linux_randombytes_getrandom'
collect2: error: ld returned 1 exit status
make[1]: *** [.././Makefile.inc:486: kyber768-CW308_STM32F3.elf] Error 1
make: *** [.././Makefile.inc:338: all] Error 2

  It seems like it didn't figure out that my OS is on linux, however I remove the #if and #elif, it can't call the function on sys like "sys/syscall.h" or "poll.h", is there anyway to solve it?

rickyhsyk, 2024-03-05:
Oh I see, I did built it with CW Makefile and use the randombytes.c from the pqlean implementation. So I got that it doesn't support STM32F303, then which one is meant for it?

rpls, 2024-03-05:
Use [this one](https://github.com/mupq/pqm4/blob/master/common/randombytes.c). You might have to adapt it, but the fallback implementation in there should be fine.

mkannwischer, 2026-08-25:
Heads-up: pqm4 is being wound down and will be archived in or after January 2027, see #409. We are unlikely to address this ourselves, but pull requests are welcome until **30 November 2026**.

vincentvbh, 2026-10-18:
This comes from building outside pqm4 with PQClean's randombytes.c, which
needs a hosted OS. Within pqm4, common/randombytes.c covers the STM32F303
with its non-random fallback, as noted above. Closing.

Drafted with Claude Opus 5.5.

closed 2026-10-18 by vincentvbh