mupq/pqm4 #336: Region 'ram' overflowed by 419416 (!) bytes

digneludvig, 2024-03-25:
Im trying to build the library with `make -j2 PLATFORM=nucleo-l476rg`. This eventually results in the following errors: 

```
/usr/lib/gcc/arm-none-eabi/10.3.1/../../../arm-none-eabi/bin/ld: elf/crypto_sign_ov-Ip_m4f_stack.elf section `.bss' will not fit in region `ram'
/usr/lib/gcc/arm-none-eabi/10.3.1/../../../arm-none-eabi/bin/ld: region `ram' overflowed by 419416 bytes
collect2: error: ld returned 1 exit status
make: *** [mupq/mk/schemes.mk:148: elf/crypto_sign_ov-Ip_m4f_stack.elf] Error 1
make: *** Waiting for unfinished jobs....
```

As you saw, the board I'm using is a Nucleo-L476RG, which is said to be a compatible board in the README, but a 419416 byte overflow seem like an absurd overflow for any board. Any ideas what is going on here and what to do about it?

mkannwischer, 2024-03-25:
OV-Ip should not run on that board because it OV needs more than 500 KiB of RAM and the board only has 128 KiB of RAM. Let me add that to the `EXCLUDED_SCHEMES` in https://github.com/mupq/pqm4/blob/master/mk/nucleo-l476rg.mk.
brb.

mkannwischer, 2024-03-25:
https://github.com/mupq/pqm4/commit/82c6d7aea48b511b7e0ac10523cadcc0aa6a1676 should fix it.
Can you try again please?

digneludvig, 2024-03-26:
> [82c6d7a](https://github.com/mupq/pqm4/commit/82c6d7aea48b511b7e0ac10523cadcc0aa6a1676) should fix it. Can you try again please?

It fixed the previous issue it seems, but instead I get the overflow below:
```
/usr/lib/gcc/arm-none-eabi/10.3.1/../../../arm-none-eabi/bin/ld: elf/mupq_crypto_sign_snova-37-8-16-4-esk_ref_stack.elf section `.bss' will not fit in region `ram'
/usr/lib/gcc/arm-none-eabi/10.3.1/../../../arm-none-eabi/bin/ld: region `ram' overflowed by 35192 bytes
collect2: error: ld returned 1 exit status
make: *** [mupq/mk/schemes.mk:148: elf/mupq_crypto_sign_snova-37-8-16-4-esk_ref_stack.elf] Error 1
make: *** Waiting for unfinished jobs....
```

digneludvig, 2024-03-26:
Thanks, it builds now, although I do get a bunch of warnings during the build process, often related to unused parameters or comparisons of expressions. These may not pose issues though?

mkannwischer, 2024-03-26:
Thanks!
Yes, these are warnings in the implementations written by other people. I'm not going to fix all these as it's way too many.

closed 2024-03-26 by mkannwischer