mupq/pqm4 #341: IndexError: index out of range

a-lve, 2024-04-16:
When I run python3 benchmarks.py,I get the following error, how can I solve it?
2024-04-16 17:31:18,689 BoardTestCase ERROR: Running kyber768 - m4fspeed - stack failed with exception: Traceback (most recent call last):                                                                 

  File "/home/lll/pqm4/mupq/mupq.py", line 239, in run_test
    output = self.interface.run(binary, self.iterations)

  File "/home/lll/pqm4/mupq/platforms.py", line 100, in run
    if self._dev.read_until(b'=')[-1] != b'='[0]:

IndexError: index out of range

kyber768 - m4fspeed FAILED                                                                                                                                                                                 
stack:   0%|                                                                                                                                                   | 0/78 [00:02<?, ?it/s, kyber768 - m4fspeed]

mkannwischer, 2024-04-17:
Can you please also post the contents of the log written by it (mupq.log)?

a-lve, 2024-04-17:
It's strange that after running the above command many times, multiple new documents are generated, such as mupq.log.4 and mupq.log.5, but some of the documents are empty, and some of them are shown as follows:2024-04-16 17:31:15,718 BoardTestCase INFO: Benchmarking kyber768 - m4fspeed
2024-04-16 17:31:15,718 BoardTestCase INFO: Runnning kyber768 - m4fspeed - stack
2024-04-16 17:31:15,718 Implementation INFO: Building kyber768 - m4fspeed - stack
2024-04-16 17:31:15,883 Implementation DEBUG: make stdout output:
make: “bin/crypto_kem_kyber768_m4fspeed_stack.bin”已是最新。
2024-04-16 17:31:18,689 BoardTestCase ERROR: Running kyber768 - m4fspeed - stack failed with exception: Traceback (most recent call last):

  File "/home/lll/pqm4/mupq/mupq.py", line 239, in run_test
    output = self.interface.run(binary, self.iterations)

  File "/home/lll/pqm4/mupq/platforms.py", line 100, in run
    if self._dev.read_until(b'=')[-1] != b'='[0]:

IndexError: index out of range
In addition, the model number of the development board I use is STM32F429ZI, not STM32F4Discovery, I don't know if it will also affect this

jelliyjane, 2024-05-17:
I am encountering the same error. If you have resolved this error, could you please share the solution?
I also use STM32F4Discovery board.

a-lve, 2024-05-27:
Hello, I encountered this problem at that time because python3 on ubuntu system reported a lot of dpkg warnings, I reinstalled the system and reinstalled python3 and it ran successfully.

mkannwischer, 2026-08-25:
Heads-up: pqm4 is being wound down and will be archived in or after January 2027, see #409. We are unlikely to address this ourselves, but pull requests are welcome until **30 November 2026**.

vincentvbh, 2026-09-12:
It's unclear how to resolve this. There isn't sufficient information about the testing environment, and it is unclear whether this issue remains in the current state.

closed 2026-10-17 by vincentvbh