mupq/pqm4 #343: Build failed

hungrydonkey, 2024-06-05:
# Env
OS: Manjaro  x86_64 6.9.2-1-MANJARO
arm-none-eabi-gcc: 14.1.0

# Compilation error
The output of  `make -j4 PLATFORM=stm32f4discovery`:
```
GEN     obj/.config.mk
make -C /home/sca/Program/projects/pqm4/libopencm3 lib/stm32/f4
  GENHDR  stm32/f3
  GENHDR  stm32/f1
  GENHDR  stm32/f0
  GENHDR  stm32/f2
  GENHDR  stm32/f4
  GENHDR  stm32/f7
  GENHDR  stm32/l0
  GENHDR  stm32/l1
  GENHDR  stm32/l4
  GENHDR  stm32/g0
  GENHDR  stm32/g4
  GENHDR  stm32/h7
  GENHDR  gd32/f1x0
  GENHDR  lpc13xx
  GENHDR  lpc17xx
  GENHDR  lpc43xx/m4
  GENHDR  lpc43xx/m0
  GENHDR  lm3s
  GENHDR  msp432/e4
  GENHDR  efm32/tg
  GENHDR  efm32/g
  GENHDR  efm32/lg
  GENHDR  efm32/gg
  GENHDR  efm32/hg
  GENHDR  efm32/wg
  GENHDR  efm32/ezr32wg
  GENHDR  nrf/51
  GENHDR  nrf/52
  GENHDR  sam/3a
  GENHDR  sam/3n
  GENHDR  sam/3s
  GENHDR  sam/3u
  GENHDR  sam/3x
  GENHDR  sam/4l
  GENHDR  sam/d
  GENHDR  vf6xx
  GENHDR  swm050
  GENHDR  pac55xx
  BUILD   lib/stm32/f4
  CC      adc_common_v1.c
  CC      adc_common_v1_multi.c
  CC      can.c
  CC      adc_common_f47.c
  CC      crc_common_all.c
  CC      crypto_common_f24.c
  CC      crypto.c
  CC      dac_common_all.c
  CC      dac_common_v1.c
  CC      dcmi_common_f47.c
  CC      desig_common_all.c
  CC      desig_common_v1.c
  CC      dma_common_f24.c
  CC      dma2d_common_f47.c
  CC      dsi_common_f47.c
  CC      exti_common_all.c
  CC      flash.c
  CC      flash_common_all.c
  CC      flash_common_f.c
  CC      flash_common_f24.c
  CC      flash_common_idcache.c
  CC      fmc_common_f47.c
[... 288 lines cut ...]
      |                            ^~~
crypto_sign/dilithium2/m4fstack/stack.h:39:24: note: expected 'uint8_t *' {aka 'unsigned char *'} but argument is of type 'const uint8_t *' {aka 'const unsigned char *'}
   39 |                uint8_t key[SEEDBYTES],
      |                ~~~~~~~~^~~~~~~~~~~~~~
crypto_sign/dilithium2/m4fstack/sign.c: In function 'crypto_sign_verify':
crypto_sign/dilithium2/m4fstack/sign.c:346:24: error: initialization of 'uint8_t *' {aka 'unsigned char *'} from incompatible pointer type 'uint8_t (*)[192]' {aka 'unsigned char (*)[192]'} [-Wincompatible-pointer-types]
  346 |   uint8_t *w1_packed = &w1_packed_comp.w1_packed;
      |                        ^
crypto_sign/dilithium2/m4fstack/sign.c:347:21: error: initialization of 'uint8_t *' {aka 'unsigned char *'} from incompatible pointer type 'uint8_t (*)[768]' {aka 'unsigned char (*)[768]'} [-Wincompatible-pointer-types]
  347 |   uint8_t *wcomp  = &w1_packed_comp.wcomp;
      |                     ^
crypto_sign/dilithium2/m4fstack/sign.c:353:20: error: initialization of 'uint8_t *' {aka 'unsigned char *'} from incompatible pointer type 'uint8_t (*)[68]' {aka 'unsigned char (*)[68]'} [-Wincompatible-pointer-types]
  353 |   uint8_t *ccomp = &ccomp_mu.ccomp;
      |                    ^
crypto_sign/dilithium2/m4fstack/sign.c:354:18: error: initialization of 'uint8_t *' {aka 'unsigned char *'} from incompatible pointer type 'uint8_t (*)[64]' {aka 'unsigned char (*)[64]'} [-Wincompatible-pointer-types]
  354 |   uint8_t *mu  = &ccomp_mu.mu;
      |                  ^
crypto_sign/dilithium2/m4fstack/sign.c:364:26: error: initialization of 'uint8_t *' {aka 'unsigned char *'} from incompatible pointer type 'uint8_t (*)[80]' {aka 'unsigned char (*)[80]'} [-Wincompatible-pointer-types]
  364 |   uint8_t *hint_ones   = &shake_hint.hint_ones;
      |                          ^
crypto_sign/dilithium2/m4fstack/sign.c:366:26: error: initialization of 'uint8_t *' {aka 'unsigned char *'} from incompatible pointer type 'uint8_t (*)[32]' {aka 'unsigned char (*)[32]'} [-Wincompatible-pointer-types]
  366 |   uint8_t *c2          = &shake_hint.c2;
      |                          ^
make: *** [mupq/mk/schemes.mk:149: elf/crypto_sign_dilithium2_m4fstack_stack.elf] Error 1
```

rpls, 2024-06-05:
I guess that might be a compiler issue, we haven't tested things with GCC 14. Maybe its more strict with pointer types.

hungrydonkey, 2024-06-06:
Adding `-Wno-incompatible-pointer-types` to the `mupq/mk/config.mk` solves this problem. A better way is to improve the source code of dilithium_m4fstack.

mkannwischer, 2024-06-13:
We are commonly testing and benchmarking with the [GNU Arm toolchain](https://developer.arm.com/downloads/-/arm-gnu-toolchain-downloads). They haven't released the gcc14 toolchain yet, so we haven't yet tested it. 
I briefly tried it with arm-none-eabi-gcc 14.1.0 from the arch package manger https://archlinux.org/packages/extra/x86_64/arm-none-eabi-gcc/, but that causes other problems that I don't have time to look into right now.
I guess we'll wait until Arm releases the new toolchain later this year. 

Feel free to open a PR to change the Dilithium sources and we'll happily merge it.

closed 2024-06-13 by mkannwischer