mupq/pqm4 #352: How to get DiLithium's pure C code optimization, so that the stack usage can reach less than 3KB

a-lve, 2024-08-12:
As mentioned in the https://eprint.iacr.org/2022/323.pdf article, the paper uses pure c optimization, so that the stack usage reaches less than 3kb, and shows that We integrated our implementation into a local fork of the benchmarking framework pqm4(Commit hash e47864b3, forked on 8 Oct 2021), but I didn't find the relevant commit and its pure C code optimization, only commit#340, but there is specific optimization code for the m4 architecture in it, but I want to use the optimization code implemented in pure C, is there any way to get it

mkannwischer, 2024-08-12:
Unfortunately, they did not publish their code - NXP never does.

If you need it to be pure C, best you can do it take our M4 optimized code and replace all assembly yourself. Should not be hard to do. 

It would be useful if we could have such an implementation in pqm4/mupq. Please open a PR if you wrote one.

mkannwischer, 2024-08-13:
@dop-amin do happen to have one lying around? This is probably something we could upstream to the reference implementation repo.

dop-amin, 2024-08-15:
> @dop-amin do happen to have one lying around? This is probably something we could upstream to the reference implementation repo.

Sorry, right now I don't have such code. Do you mean just *any* C code or C code implementing common optimizations?

mkannwischer, 2024-08-15:
I think having anything with an optimized memory footprint would be good. Speed optimizations can be added later if needed.

dop-amin, 2024-09-26:
@a-lve I removed the parts that had been written assembly for M4. You can find my fork of the pq-crystals repo including memory optimization here https://github.com/dop-amin/dilithium/tree/lowram. It's also submitted as a PR to pq-crystals pq-crystals/dilithium#91. 
Hope that helps!

ate-wang, 2025-01-22:
I want to know if anyone has tested the performance of Kyber and Dilithium in the M4 environment, and what can be achieved in the M4 environment? This includes key pair generation, encapsulation, decapsulation, signing, signature verification, etc. Has anyone done relevant tests that you can share with me?

mkannwischer, 2026-08-25:
Heads-up: pqm4 is being wound down and will be archived in or after January 2027, see #409. We are unlikely to address this ourselves, but pull requests are welcome until **30 November 2026**.

vincentvbh, 2026-09-12:
I'm closing this one. I'll handle stack-optimized Dilithium in another issue.

closed 2026-09-12 by vincentvbh