mupq/pqm4 #359: Add context info to benchmark.md

franslundberg, 2024-09-27:
Great to see the benchmark results in benchmark.md. Would be great to have some more info on how the tests where performed. On what platform and with what compiler settings. For example, for flash size, was the -Os flag used? Not super-easy to find out by looking at the code that generated it. At least not easy for me.

rpls, 2024-09-27:
Good idea actually. The info is somewhat scattered throughout the `README.md`, and not up-to-date. I'll think about a way on how to automatically include this with the scripts.

FYI: benchmarks are created with the `benchmark.py` script without any additional optimization flags to the script. For `pqm4`, this results in compilation with `-O3`. By default, the compilation is done in one translation unit, so similar to, but not quite, LTO. For cycle count-based benchmarks, the target boards are clocked low-enough, that the flash doesn't need any wait-states. Default board right now is the Nucleo-L4R5ZI. The used compiler changes but we aim to use the "official" GNU GCC Toolchain provided by ARM.

franslundberg, 2024-09-30:
Thanks for the info. Much appreciated.