mupq/pqm4 #374: Build failed for cw308t-stm32f3

nlcruise, 2025-01-13:
When building the current master branch (latest commit [49ce5be](https://github.com/mupq/pqm4/commit/49ce5bea56c2a00da2671a54949d9f214936ca21)), the building process stops:
```
$ make -j9 PLATFORM=cw308t-stm32f3
(...)
/usr/lib/gcc/arm-none-eabi/10.3.1/../../../arm-none-eabi/bin/ld: elf/mupq_crypto_sign_cross-sha2-r-sdpg-5-fast_ref_stack.elf section `.bss' will not fit in region `ram'
/usr/lib/gcc/arm-none-eabi/10.3.1/../../../arm-none-eabi/bin/ld: region `ram' overflowed by 4 bytes
collect2: error: ld returned 1 exit status
make: *** [mupq/mk/schemes.mk:149: elf/mupq_crypto_sign_cross-sha2-r-sdpg-5-fast_ref_stack.elf] Error 1
make: *** Waiting for unfinished jobs....

```
Even after changing the RAM-values in libopencm3/ld/devices.data to 64K (while the device has only 40KB RAM), it gives the same error. I want to use PQM4 to test FrodoKEM.

rpls, 2025-01-13:
I'll check it out in the next few days, but for a quick fix you could do the following:

```
diff --git a/mk/cw308t-stm32f3.mk b/mk/cw308t-stm32f3.mk
index c08797a..6f0f439 100644
--- a/mk/cw308t-stm32f3.mk
+++ b/mk/cw308t-stm32f3.mk
@@ -27,6 +27,8 @@ EXCLUDED_SCHEMES = \
 	mupq/crypto_sign/cross-sha3-r-sdp-5-small% \
 	mupq/crypto_sign/cross-sha2-r-sdp-5-fast% \
 	mupq/crypto_sign/cross-sha3-r-sdp-5-fast% \
+	mupq/crypto_sign/cross-sha2-r-sdpg-5-fast% \
+	mupq/crypto_sign/cross-sha3-r-sdpg-5-fast% \
 	mupq/crypto_sign/meds55604% \
 	mupq/crypto_sign/meds41711% \
 	mupq/crypto_sign/meds167717% \
```

That will exclude cross from building on the platform. Add any other offending schemes there as well, if any more pop up.

But as for FrodoKEM: before we removed it, it used up to 80KiB of Memory. So unless you use a newer memory-efficient implementation, FrodoKEM won't run on the STM32F3.

rpls, 2025-01-13:
And use the newest [GNU ARM Toolchain](https://developer.arm.com/downloads/-/arm-gnu-toolchain-downloads) (one for `arm-none-eabi`).