mupq/pqm4 #389: assembly NTT implementation doest return same values of the vanilla implementation of Dilithium/ML-DSA

kuolemaaa, 2025-02-14:
assembly NTT implementation does not return same values of the vanilla implementation of Dilithium/ML-DSA.

What I tried:
On stm32cubeide flashed on a nucleo f411re
 - generate an array of lenght 256 of int32 filled with numbers 0 to 255 
 - apply the ntt implementation of pqm4 to the array
```
  int32_t a[N];
  int32_t a_star[N];

  for(int i=0; i<N; i++){
	  a[i]=i;
	  a_star[i]=i;
  }

  ntt(a_star);
```
First 10 values I got:
```
a_star[0]	int32_t	-356594	
a_star[1]	int32_t	4949942	
a_star[2]	int32_t	13884114	
a_star[3]	int32_t	7227518	
a_star[4]	int32_t	-4303253	
a_star[5]	int32_t	903461	
a_star[6]	int32_t	2287113	
a_star[7]	int32_t	3389395	
a_star[8]	int32_t	9828353	
a_star[9]	int32_t	3912035	
a_star[10]	int32_t	3833152	
```

 on x86 computer:
- generate an array of lenght 256  of int32 filled with numbers 0 to 255 
- apply the ntt from the reference code of dilithium2/ml-dsa (they are the same iirc)
```
    int32_t a[N];
    int32_t a_star[N];
    int32_t a_recved[N];

    for (int i = 0; i < N; i++) {
        a[i] = i;
        a_star[N]=a[i];
    }

    //print_array(a);

    ntt(a_star);
```

Ref code NTT func at https://github.com/pq-crystals/dilithium/blob/master/ref/ntt.c

First 10 values I got in normal and bitreversed mode:
```
[00000000][   0]        -3044312  --  [00000000][   0]        -3044312
[00000001][   1]        -5448616  --  [10000000][ 128]        -2330864
[00000010][   2]       -13133036  --  [01000000][  64]        11810295
[00000011][   3]       -11023096  --  [11000000][ 192]       -13142360
[00000100][   4]        -5742915  --  [00100000][  32]        15563313
[00000101][   5]         2020873  --  [10100000][ 160]         3980689
[00000110][   6]        -2561512  --  [01100000][  96]        -3457219
[00000111][   7]        -9649826  --  [11100000][ 224]        -4014340
[00001000][   8]       -19254547  --  [00010000][  16]       -11317290
[00001001][   9]       -14648991  --  [10010000][ 144]         7826615
[00001010][  10]        -7063034  --  [01010000][  80]         7472076
```

it seems that not a single value is the same. is it normal?

mkannwischer, 2025-02-15:
No, values should be the same modulo q - but they are not in your test.

```
  for (int i = 0; i < N; i++) {
        a[i] = i;
        a_star[N]=a[i];
    }
```
looks wrong, you are only setting `a_star[N]` (which is even out of bounds).

kuolemaaa, 2025-02-15:
Yea. I'm dumb and paniked. It slipped from me.

closed 2025-02-15 by kuolemaaa