mupq/pqm4 #390: Difference between matacc_opt32() and matacc_cache32()

rsk2002, 2025-03-03:
Hi,

The difference I see is that in matacc_cache32() function zetas is used, but in matacc_opt32() function zetas is not used. Can any one please help me understand the functional difference between these two functions. Also, can anyone please help me to understand the significance of zetas.

Thank you

[image]

mkannwischer, 2025-03-03:
This is a trick from https://eprint.iacr.org/2021/986 - it's called asymmetric multiplication in that paper. 
In the basemul one has to compute:
```
c0 = a1 * b1 * zeta + a0 * b1
c1 = a0 * b1 + a1 * b0
```

Now if b is used in multiple polynomial multiplications, one can compute `b1 * zeta` only once and cache the result. 
You can either do this by merging it into the NTT (as proposed in the paper), or you merge it into first use of it (that's done in the M4 code you are looking at). 

`matacc_cache32()` will write the result of `b1 * zeta` into `b_prime`, while `matacc_opt32` consumes it from `b_prime` and hence does not need any zetas.

Does that make sense?

vincentvbh, 2025-03-26:
To clarify, this is the small-dimensional case of Toeplitz matrix-vector product that I didn't recognize when I discovered. I already clarified that in my master's thesis.