mupq/pqm4 #395: why sphincs+ 256s version signature gen is faster than 192s version

yberlin, 2025-07-29:
[pqm4](https://github.com/mupq/pqm4/tree/master)
/benchmarks.md
sphincs-shake-192s-simple AVG: 43,114,327,277

sphincs-shake-256s-simple AVG: 38,175,697,620

mkannwischer, 2025-07-29:
Take a look at https://github.com/slh-dsa/slhdsa-c/blob/main/test/new_param.csv
This contains the number of hash invocations per parameter set (SLH-DSA, not SPHINCS+, but I don't think the number of hashes differ). 
It says 3799155 hashes for sphincs-shake-192s-simple and 3313122 hashes for sphincs-shake-256s-simple. 

Since the hashes are by far the most expensive part on M4, I'd say the pqm4 numbers do make sense?

mkannwischer, 2026-08-25:
Heads-up: pqm4 is being wound down and will be archived in or after January 2027, see #409. We are unlikely to address this ourselves, but pull requests are welcome until **30 November 2026**.

vincentvbh, 2026-09-12:
This doesn't look like an issue. And the question appears to be answered already.

closed 2026-09-12 by vincentvbh